stack-of-tasks/pinocchio #916: Tag new version

wxmerkt, 2019-10-24:
Similar to eigenpy, it would be great if we can add a new tag - there are a few important changes since 2.1.9 including the Hessians for the kinematics, the argument checking with exceptions, and the package.xml that has been added.

As part of this tag, I will deprecate the previous catkin wrapper and release directly from this repository to the ROS buildfarm.

Packages depending on `ros-[melodic/kinetic]-pinocchio` package then need to change how they include Pinocchio from `find_package(catkin REQUIRED COMPONENTS pinocchio)` to `find_package(PkgConfig) ; pkg_check_modules(pinocchio REQUIRED pinocchio)`. In other words, the ROS-buildfarm-provided version will behave exactly as the default CMake build and robotpkg version.

jcarpent, 2019-10-24:
I will do it tomorrow morning. Small question, did you manage to get ride from the issues your mentioned in #914? In other words, is it still a priority for you?

wxmerkt, 2019-10-25:
We can work with the PkgConfig for now but CMake as in #914 would still be preferable - it is not a priority and not blocking for the new tag though.

jcarpent, 2019-10-25:
Done. Thanks a lot @wxmerkt for pushing the ROS integration.

closed 2019-10-25 by jcarpent

wxmerkt, 2019-10-25:
Thank you @jcarpent :-)

wxmerkt, 2019-10-25:
I realised we didnt update the package version in the package.xml to 2.1.10. I just added a commit to update it to 2.1.10. Is it okay to re-tag to https://github.com/stack-of-tasks/pinocchio/commit/6476dcf48691e28e0c31e0aad185ebe544efdbac?

nim65s, 2019-10-25:
If you are speaking about removing a published tag, and then tagging another commit with the same name, I would really like that we don't do that.

But we could make another, like v2.1.11

wxmerkt, 2019-10-25:
I don't think that tag has seen any use since it was just made this morning. Happy to do 2.1.11 if we before update the package.xml

jcarpent, 2019-10-27:
@wxmerkt I did it. Is it fine now?

wxmerkt, 2019-10-27:
Thank you, perfect :-)