stack-of-tasks/pinocchio #923: hide mimicJoints in model

alexswerner, 2019-10-31:
Hi,

when importing a model with buildModelFromUrdf, i get all joints which are marked as mimicJoints listed as joints. This is e.g. used in the romeo model distributed with pinocchio, typically for coupled finger joints. Is there a way to remove these joints/links from the model to (a) reduce computation time (b) make the number of joints and actuators match. I think that would simplify development. A function like:
model.pruneSubtree(joint_idx) would be a possible solution.

Cheers,
Alex

jcarpent, 2019-10-31:
Hi @alexswerner,

Your request is very similar to the one from @markusgft in #905. If you confirm, I will try to implement it tomorrow. Is it fine for you?

alexswerner, 2019-11-01:
Great, i will comment on #905. Thanks

jcarpent, 2019-11-01:
Duplicate with #905.

closed 2019-11-01 by jcarpent