stack-of-tasks/pinocchio #934: Speeding Up Computation Time of Inverse Dynamics

gsutanto, 2019-11-06:
Dear Pinocchio Authors,

I am using Pinocchio's RNEA inverse dynamics computation in a real-time control loop at 1000 Hz, which means 1ms control window. However, I measured the RNEA computation time itself is 0.5ms. Is there anyway to make it compute faster?

Best Regards,

Giovanni Sutanto

gabrielebndn, 2019-11-06:
Normal execution times are in the order of a few microseconds.
Are you sure you are compiling your project with the correct compilation flags? `-O3` should give best performances

gsutanto, 2019-11-06:
Oh yeah, you're right. Once I use Release mode and -O3 flag, the computation time goes down to 3 microseconds. Thanks a lot @gabrielebndn !

closed 2019-11-06 by gsutanto

jcarpent, 2019-11-07:
@gsutanto Just for information, what is the model of your robot? model.nq? model.nv?

gsutanto, 2019-11-07:
Hi @jcarpent , model.nq=7, model.nv=7

jcarpent, 2019-11-08:
You may have a look at the paper https://hal-laas.archives-ouvertes.fr/hal-01866228 which describes Pinocchio. And you will see some benchmarks on several robots.
You can expect to go below 1us for the RNEA. Can you run:
```
make bench
./benchmarks/timings
```
It will show the bench for a standard humanoid robot, but you can also do
```
./benchmarks/timings my_urdf_model.urdf
```
for your specific robot.

Thanks in advance for your reply.

gsutanto, 2019-11-08:
Dear @jcarpent ,

I see. Thanks for the references. For now, this is fine for our need. If we need optimize further, we will refer to (and potentially re-open) this thread.

Thanks!

Best Regards,

Giovanni