stack-of-tasks/pinocchio #939: updating only the momentum quantities.

proyan, 2019-11-08:
We need a function in crocoddyl, that updates the centroidal momentum hg, without additional computations Currently the version that we have (computeCentroidalDynamics) computes the kinematics as well, and thus is not as efficient.

I'm willing to implement it here, but I wish to affirm that there is an appetite for such a function in Pinocchio. What do you think?

jcarpent, 2019-11-08:
This is fine for me. I can also handle it before doing a new release. What do you think?

cmastalli, 2019-11-08:
There is two ways of doing, I'm not sure what it's the Pinocchio way of doing:
 1. `updateCentroidalDynamics(model, data)`
 2. `computeCentroidalDynamics(model, data) #we don't pass q and qd`

Which option should we pursue @jcarpent ?

jcarpent, 2019-11-08:
I think the best naming is `updateCentroidalDynamics` as done in other places in Pinocchio.

proyan, 2019-11-08:
If you can handle it, I'm not going to say no :smiley:

gabrielebndn, 2019-11-08:
My bit: I think `computeCentroidalDynamics(model,data)`. Most methods (think of `forwardDynamics`) can work with or without the configuration.
This way we could establish a nice pattern in all methods: if the configuration is passed, then it is used to compute the kinematics, otherwise the kinematics is not computed

jcarpent, 2019-11-08:
@gabrielebndn Not for `forwardDynamics`, because you don't know if you want the zero, first or second kinematics. The same applied for `computeCentroidalDynamics`. One has to understand the update by `update from the current value in Data`.

cmastalli, 2019-11-08:
> If you can handle it, I'm not going to say no smiley

Unfortunately, I don't have time. Thanks for making this effort @proyan :smiley:

jcarpent, 2019-11-08:
@cmastalli Rohan was talking to me ;)

proyan, 2019-11-08:
@jcarpent Worth the try :D
No problem, I'll make the PR within next week. I'll prefer `updateCentroidalDynamics`.
By the way, I don't think that I can do it with simply `updateCentroidalDynamics(model, data)`. Correct me if I am wrong, we'll need `v` in order to update the centroidal dynamics. And just inputting `v`, without `q` seems bad.

gabrielebndn, 2019-11-08:
For `forwardDynamics` and the like, we assume that the forward kinematics has been computed up to the first time derivative. So I do not think you need to take `v` into account. As long as the `data.v[i]` are filled with the correct values, I do not think you need anything else

jcarpent, 2019-11-08:
The answer from @gabrielebndn is correct. But, I was thinking to add a enum to say that we want:
`computeCentroidalDynamics(model,data,q,v)` maching `computeCentroidalDynamics(model,data,FIRST_ORDER)` and `computeCentroidalDynamics(model,data,q,v,a)` matching `computeCentroidalDynamics(model,data,SECOND_ORDER)`.

What do you think?

gabrielebndn, 2019-11-09:
@jcarpent ooh, now I understand the issue... You have two computeCentroidalDynamics, one being first order and the other one second order, but without the arguments you cannot tell which one you want...
In my opinion, I've always found the name computeCentroidalDynamics to be confusing... You could rename them computeCentroidalMomentum and computeCentroidalMomentumTimeDerivative, it would be more informative and there would be no ambiguity when no arguments are provided...

jcarpent, 2019-11-10:
Sounds a good idea indeed. I will provide such functions.

proyan, 2019-11-10:
I agree with @gabrielebndn, centroidalDynamics is quite vague.
So I create the new function with `computeCentroidalMomentum(model, data)`, and deprecate and rename the original functions?

jcarpent, 2019-11-10:
As I mentioned above, I will take in charge @proyan. I do apply some other modifications elsewhere.

jcarpent, 2019-11-10:
@proyan But, if it is urgent for you, please go ahead, I will adjust my code accordingly.

jcarpent, 2019-11-10:
Anyway, I think you can keep the current functions, but the two signatures you need:
```
computeCentroidalMomentum(model,data)
computeCentroidalMomentumTimeDerivatives(model,data)
```
because
```
computeCentroidalDynamics(model,data,q,v)
computeCentroidalDynamics(model,data,q,v,a)
```
are clear enough I think, no?

We may add shortcut with the naming proposed by @gabrielebndn.
@proyan @gabrielebndn @cmastalli What do you think?

jcarpent, 2019-11-10:
After thinking a bit, it will too much redundant to cope these namings. So, it would nice indeed to mark the former naming as deprecated.
Sorry for the noise ;)

gabrielebndn, 2019-11-10:
I think `computeCentroidalDynamics` is a bit confusing. I suggest imoplementing the four signatures
```
computeCentroidalMomentum(model,data)
computeCentroidalMomentumTimeDerivative(model,data)
computeCentroidalMomentum(model,data,q,v)
computeCentroidalMomentumTimeDerivative(model,data,q,v,a)
```
and deprecating
```
computeCentroidalDynamics(model,data,q,v)
computeCentroidalDynamics(model,data,q,v,a)
```
which will just call `computeCentroidalMomentum` and `computeCentroidalMomentumTimeDerivative`, respectively. It is better if the methods keep the same names in the version with and without arguments

jcarpent, 2019-11-12:
Solved by #944.

closed 2019-11-12 by jcarpent