stack-of-tasks/pinocchio #940: pc (pkg-config) file missing space around ">=" sign

andreadelprete, 2019-11-08:
Installing the latest version of pinocchio (devel) from source I got an apparently faulty pc file, which contained the following line:
```
Requires: eigen3>=3.0.5, urdfdom>=0.2.0
```
The lack of a "space" before and after ">=" gave problem to pkg-config on my machine (Ubuntu 16.04). For instance:
```
$ pkg-config --print-requires pinocchio
Package eigen3>=3.0.5 was not found in the pkg-config search path.
Perhaps you should add the directory containing `eigen3>=3.0.5.pc'
to the PKG_CONFIG_PATH environment variable
Package 'eigen3>=3.0.5', required by 'pinocchio', not found
```
Adding the missing spaces solved the problem.

nim65s, 2019-11-08:
Hi @andreadelprete,

This has already been reported by @wxmerkt  here: https://github.com/stack-of-tasks/pinocchio/pull/933#issuecomment-551333575 ; and @sbarthelemy  is on it.

nim65s, 2019-11-08:
You can just temporarily and locally revert 2adf778 to fix this, if necessary

andreadelprete, 2019-11-08:
ok thanks!

jcarpent, 2019-11-08:
@andreadelprete https://github.com/jrl-umi3218/jrl-cmakemodules/pull/278 should fix the current issue.

jcarpent, 2019-11-08:
@sbarthelemy I have proposed a fix because it is affecting current users. Sorry for the bypass.

jcarpent, 2019-11-08:
Solved by #941.

closed 2019-11-08 by jcarpent