stack-of-tasks/pinocchio #943: python3 installation on Mac

ddliugit, 2019-11-09:
I need to upgrade python2 to python3 and tried the following command. However, the error appears.

`brew install pinocchio-python3`

Can you help me check?

error:

> 10-16-176-1:build dongdong$ brew install pinocchio-python3
> Updating Homebrew...
> ==> Auto-updated Homebrew!
> Updated 1 tap (homebrew/core).
> ==> Updated Formulae
> proj
> 
> ==> Installing pinocchio-python3 from gepetto/gepetto
> ==> Downloading https://github.com/stack-of-tasks/pinocchio/releases/download/v2
> 
> curl: (22) The requested URL returned error: 404 Not Found
> Error: Failed to download resource "pinocchio-python3"
> Download failed: https://github.com/stack-of-tasks/pinocchio/releases/download/v2.1.9/pinocchio-python3-2.1.9.mojave.bottle.tar.gz
> Warning: Bottle installation failed: building from source.
> ==> Downloading https://github.com/stack-of-tasks/pinocchio/releases/download/v2
> Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/be973ed7fc613bc57e003f3310ffb28c46121d3e6832bb547a6e8825abff8d74--pinocchio-2.1.9.tar.gz
> ==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
> ==> make
> Last 15 lines from /Users/dongdong/Library/Logs/Homebrew/pinocchio-python3/02.make:
> /usr/local/include/assimp/LogStream.hpp:68:16: error: expected ';' at end of declaration list
>     LogStream() AI_NO_EXCEPT;
>                ^
> /usr/local/include/assimp/LogStream.hpp:99:24: error: expected function body after function declarator
> LogStream::LogStream() AI_NO_EXCEPT {
>                        ^
> /usr/local/include/assimp/defs.h:301:24: note: expanded from macro 'AI_NO_EXCEPT'
> #  define AI_NO_EXCEPT noexcept
>                        ^
> fatal error: too many errors emitted, stopping now [-ferror-limit=]
> 20 errors generated.
> make[2]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/parsers/expose-parsers.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> make[1]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/all] Error 2
> make: *** [all] Error 2
> 
>

jcarpent, 2019-11-09:
I will have a look at it. On which OS are you working on: 10.14 or 10.15?

ddliugit, 2019-11-09:
10.14

ddliugit, 2019-11-09:
My python version is 3.7

jcarpent, 2019-11-10:
@ddliugit The bug is coming from a new release of assimp. I will fix it asap.

jcarpent, 2019-11-10:
There is a bug in Assimp since version 5. I have proposed a fix https://github.com/assimp/assimp/pull/2758.
In the meantime, before any new release, I've also suggested a fix in hpp-fcl https://github.com/humanoid-path-planner/hpp-fcl/pull/95.
This fix has been included in the brew packages of pinpcchio's dependencies.

The mirror is now ok, you should be able to do:
`brew reinstall pinocchio-python3`.

I will close this issue as it seems to be fixed now. @ddliugit Feel free to reopen if needed.

closed 2019-11-10 by jcarpent

ddliugit, 2019-11-10:
**I just tried, but it doesn't work.**

$ brew reinstall pinocchio-python3
==> Reinstalling gepetto/gepetto/pinocchio-python3 
==> Downloading https://github.com/stack-of-tasks/pinocchio/releases/download/v2

curl: (22) The requested URL returned error: 404 Not Found
Error: Failed to download resource "pinocchio-python3"
Download failed: https://github.com/stack-of-tasks/pinocchio/releases/download/v2.1.9/pinocchio-python3-2.1.9.mojave.bottle.tar.gz
Warning: Bottle installation failed: building from source.
==> Downloading https://github.com/stack-of-tasks/pinocchio/releases/download/v2
Already downloaded: /Users/dongdong/Library/Caches/Homebrew/downloads/be973ed7fc613bc57e003f3310ffb28c46121d3e6832bb547a6e8825abff8d74--pinocchio-2.1.9.tar.gz
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG
==> make
Last 15 lines from /Users/dongdong/Library/Logs/Homebrew/pinocchio-python3/02.make:
/usr/local/include/assimp/LogStream.hpp:68:16: error: expected ';' at end of declaration list
    LogStream() AI_NO_EXCEPT;
               ^
/usr/local/include/assimp/LogStream.hpp:99:24: error: expected function body after function declarator
LogStream::LogStream() AI_NO_EXCEPT {
                       ^
/usr/local/include/assimp/defs.h:301:24: note: expanded from macro 'AI_NO_EXCEPT'
#  define AI_NO_EXCEPT noexcept
                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/parsers/expose-parsers.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [bindings/python/CMakeFiles/pinocchio_pywrap.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/gepetto/homebrew-gepetto/issues

jcarpent, 2019-11-10:
Are you up to date with brew? Try first:
```
brew update
brew uninstall pinocchio-python3
brew reinstall pinocchio-python3
```
If it does not work properly, do:
```
brew untap gepetto/gepetto
brew tap gepetto/gepetto
brew reinstall pinocchio-python3
```
Best,

Justin

ddliugit, 2019-11-10:
Now it works!

However when I try to run some examples of Pinocchio, it doesn't work!

:wsmemmo_pinocchio dongdong$ python arm3d.py 
Traceback (most recent call last):
  File "arm3d.py", line 1, in <module>
    from robots import loadTalosArm
  File "/Users/dongdong/repos/wsmemmo_pinocchio/robots.py", line 2, in <module>
    from pinocchio.utils import *
ModuleNotFoundError: No module named 'pinocchio.utils'

ddliugit, 2019-11-10:
Did you change the module? or should I change the PYTHONPATH in the bash_profile?

ddliugit, 2019-11-10:
import pinocchio as pin
from pinocchio.utils import *

print(eye(6))

error:

  File "5.py", line 8, in <module>
    print(eye(6))
NameError: name 'eye' is not defined

jcarpent, 2019-11-10:
Yes, you should change the PYTHONPATH to something like:
```
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.7/site-packages
```

ddliugit, 2019-11-10:
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.7/site-packages
export PATH="/opt/local/bin:/opt/local/sbin:$PATH"
export PATH=$PATH:/Users/dongdong/bin
export PATH=/usr/local/bin:$PATH
export PYTHONPATH=/opt/drake/lib/python3.7/site-packages:${PYTHONPATH}
export PATH=/usr/local/opt/python/libexec/bin:${PATH}

ddliugit, 2019-11-10:
I have already added it to PYTHONPATH though

ddliugit, 2019-11-10:
I check the file inside the /usr/local/lib/python3.7/site-packages/pinocchio
I can only find the visualize folder inside and it is empty.
The software is not installed to the right directory?

jcarpent, 2019-11-10:
it works on my computer.
So, it seems it come from your side. Could do something like:
```
brew unlink pinocchio
brew link pinocchio-python3
```

ddliugit, 2019-11-10:
OK. Now the error changes:

>>> from pinocchio.utils import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/pinocchio/__init__.py", line 6, in <module>
    from .robot_wrapper import RobotWrapper
  File "/usr/local/lib/python3.7/site-packages/pinocchio/robot_wrapper.py", line 5, in <module>
    from . import libpinocchio_pywrap as pin
ImportError: dlopen(/usr/local/lib/python3.7/site-packages/pinocchio/libpinocchio_pywrap.so, 2): Library not loaded: /usr/local/opt/assimp/lib/libassimp.4.dylib
  Referenced from: /usr/local/opt/hpp-fcl/lib/libhpp-fcl.dylib
  Reason: image not found

ddliugit, 2019-11-11:
It works after reinstall hpp-fcl!