stack-of-tasks/pinocchio #947: robotpkg binary does not depend on libeigen3-dev

wxmerkt, 2019-11-12:
Similar to https://github.com/stack-of-tasks/eigenpy/issues/112

jcarpent, 2019-11-12:
Thanks a lot @wxmerkt for the issue. I hope it will be fixed soon.

jcarpent, 2019-11-13:
I think all the discussions are discussed there: stack-of-tasks/eigenpy#112. I will close this issue.
Thanks @wxmerkt for raising it.

closed 2019-11-13 by jcarpent

wxmerkt, 2019-11-13:
Has it been addressed in a new release though? I don't think we should close issues until they have been fixed

jcarpent, 2019-11-13:
Yes, it will be handled for the next release, following the discussions you have in stack-of-tasks/eigenpy#112. The release process of the Pinocchio project is independent of the binaries delivered by robotpkg. But if you think important to reopen it, feel free to do it. But I prefer to not have similar channels of discussions on a similar topic.