stack-of-tasks/pinocchio #949: Addind dDifference function for multi-joints

cmastalli, 2019-11-13:
Dear all,

It would be nice to have this function in Pinocchio. Currently, I am retrieving these Jacobians from the dIntegrate in Crocoddyl. However, I believe that making a full matrix factorization is very inefficient. Therefore, this task has to be done at the Pinocchio level.

Unfortunately, I don't fully understand the Pinocchio implementation. And it would take me a lot of time, more than available, to develop such kind of function. 

Who could deal with it? @jmirabel, @gabrielebndn or @jcarpent 

I would be eternally grateful with the person that makes this a reality :smiley: 

Yours,
Carlos

jcarpent, 2019-11-13:
I can handle it, but I don't know when exactly.

cmastalli, 2019-11-13:
This is not urgent, since I have provided an inefficient solution through full factorization.

However, whenever, it is implemented. Could you post it here? So I can track changes and updates Crocoddyl.

Merci

jcarpent, 2019-11-14:
fixed by #950.

closed 2019-11-14 by jcarpent