stack-of-tasks/pinocchio #955: Can pinocchio calculate the minimum distance between two meshes

ddliugit, 2019-11-15:
I am working on an optimization problem which need a kinematic constraint of calculating the minimum distance between two meshes.

Can I use _pinoccchio_ to get the minimum distance between two meshes for a certain configuration `q`?

like so:

`x = minimumdistance(robot,q,partId1, partId2)`

jmirabel, 2019-11-15:
You have *computeDistances(model, data, geomModel, geomData, q)*. Then you have to find the minimum distance yourself from the distance results (geomData.distanceResults).

Mind that you will have to filter the collision pairs in order to have relevant results.

ddliugit, 2019-11-15:
Thanks for your timely reply!

I check the function and find cpp version of that function. I am using python right now and do you have any plan to add that function in python version?

Another issue is that I also need to calculate the derivative of distance with respect to robot configuration, would that be possible in Pinocchio?

jmirabel, 2019-11-15:
If you have the source, it shouldn't be hard to expose computeDistances.

The DistanceResult objects of hpp-fcl, in geomData.distanceResults, contains all the info you need to compute that derivative. I doubt the algorithm is implemented in Pinocchio but it isn't hard to write.

jcarpent, 2019-11-15:
For the second question, is just related to the Jacobian of the witness points, so its very easy to write by hand ;) I do not think that a dedicated function is needed currently, but we may think to add it in Future.

ddliugit, 2019-11-15:
Thanks, all.

@jcarpent I see. I just don't think too much about that. The derivative of the distance between one point on the robot and the ground can be considered the derivative of pose of that point with respect to `q`, which equals to the Jacobian of that point. 

Is that what you want to say?

gabrielebndn, 2019-11-15:
@ddliugit, `computeDistances` is already exposed in Python

ddliugit, 2019-11-15:
I see that function in expose-geometry.cpp.

Thanks for your reminder.

jcarpent, 2019-11-15:
> Thanks, all.
> 
> @jcarpent I see. I just don't think too much about that. The derivative of the distance between one point on the robot and the ground can be considered the derivative of pose of that point with respect to `q`, which equals to the Jacobian of that point.
> 
> Is that what you want to say?

Yes, this is correct. Just check by Finite differences to be sure your result is correct.

ddliugit, 2019-11-15:
Thanks all!

closed 2019-11-15 by ddliugit