stack-of-tasks/pinocchio #961: Adding aligned vector for df_dq, df_dv and df_dtau

cmastalli, 2019-11-21:
Dear @jcarpent,

I believe you will somehow release a version with the derivatives of the contact dynamics.
Then, we will need to add aligned vector (per each joint) for these derivatives inside `pinocchio::Data`. 

So, I propose to first include these data. This is a blocking point for me now because I can't access the force data inside cost functions in Crocoddyl. I would like to use to fill my custom computation of derivatives, before waiting for the contact dynamic derivatives.

I want to implement it now, is this makes sense to you?

Yours,
Carlos

jcarpent, 2019-11-21:
I don't really get which field you are talking about. Can you provide a link or an example?

cmastalli, 2019-11-21:
`df_dq`, `df_dv` and `df_dtau`, derivatives of the contact forces expressed in the parent frame.

jcarpent, 2019-11-21:
In my mind, we should not think of contact derivatives at the joint level. In other words, you might want to evaluate a frame derivative (linked to a joint), but we will never fill the corresponding joint df_dtau (because this is not needed). So, I think populating Data with these quantities will not be useful.
I would rather propose an interface where you can provide output arguments to be filled, and of course, these arguments can be stored inside another struct of your choice. Is this solution feasible for you?

cmastalli, 2019-11-21:
> In my mind, we should not think of contact derivatives at the joint level. In other words, you might want to evaluate a frame derivative (linked to a joint), but we will never fill the corresponding [...]

We could generate an empty vector that can be assigned its size according to the case. We will need this to compute the derivatives of forces as explained in Crocoddyl paper: https://cmastalli.github.io/publications/crocoddyl20unpub.pdf

> I would rather propose an interface where you can provide output arguments to be filled, and of course, these arguments can be stored inside another struct of your choice. Is this solution feasible for you?

What do you mean with interface? is an interface to `pinocchio::Data`? Could you provide an example code?

jcarpent, 2019-11-21:
> We could generate an empty vector that can be assigned its size according to the case.

As it is application dependant, I would suggest not doing that in Pinocchio, but rather in the related project by deriving from Data.

> What do you mean with interface? is an interface to `pinocchio::Data`? Could you provide an example code?

I just mean in the function signature, you can pass output arguments of the function.

jcarpent, 2019-11-22:
I will close this issue. If you need, feel free to reopen it when needed.

closed 2019-11-22 by jcarpent