stack-of-tasks/pinocchio #969: boost container limits

hammoudbilal, 2019-11-28:
I have a library with boost-bindings that depends on pinocchio, when trying to compile, 
i get the following error 

usr/local/include/pinocchio/container/boost-container-limits.hpp:29:7: error: "BOOST_MPL_LIMIT_LIST_SIZE value is lower than the value of
      PINOCCHIO_BOOST_MPL_LIMIT_CONTAINER_SIZE"
    # error "BOOST_MPL_LIMIT_LIST_SIZE value is lower than the value of PINOCCHIO_BOOST_MPL_LIMIT_CONTAINER_SIZE"

I tried adding 
#include <pinocchio/container/boost-container-limits.hpp>

but it seems to generate the same error. any idea or an example on how to handle this ?

jcarpent, 2019-11-28:
- Do you a link to your code for inspection?
- Do you use pkg-config to retrieve Pinocchio settings? 
- One quick fix is to include pinocchio before all others include:
```
#include <pinocchio/fwd.hpp>
```

hammoudbilal, 2019-11-28:
yes I use pkg-config, the solution was to move #include <boost/python.hpp> to be the last one. Thanks for the help

closed 2019-11-28 by hammoudbilal

jmirabel, 2020-02-10:
Hi @jcarpent 

I keep having this problem and the only solution that works is the inclusion of pinocchio/fwd.hh. Only using the compilation flags from the pc file does not work. Does it work for you ?

jmirabel, 2020-02-10:
I have boost 1.58, on Ubuntu 16.04.

jmirabel, 2020-02-11:
And this is not the case on 18.04.

GithubMingEnter, 2024-08-27:
I have include fwd.hpp, but it hints same problem