stack-of-tasks/pinocchio #970: Runtime Eigen assertion (TensorStorage.h) on 16.04 in Debug mode

Leph, 2019-11-28:
I am facing a runtime assertion from Eigen related to Tensor on 16.04, both with current master and devel branch.
This assersion is not showing up when compiling in Release (-DNDEBUG) mode.
In Release mode, Pinocchio is running fine.
Note that the unittests are fine as well, even when compiled in Debuig mode.

Here is a way to reproduce:

```
#include <pinocchio/multibody/model.hpp>
#include <pinocchio/multibody/data.hpp>
#include <iostream>

int main()
{
    pinocchio::Model _model;
    std::cout << "Eigen version: " << EIGEN_WORLD_VERSION << "." << EIGEN_MAJOR_VERSION << "." << EIGEN_MINOR_VERSION << std::endl;
    pinocchio::Data _data(_model);
    
    return 0;
}
```

Compiling command:
`g++ -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DBOOST_MPL_LIMIT_VECTOR_SIZE=30 -DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_WITH_URDFDOM -std=c++11 -W -Wall -I /tmp/pinocchio/build/include/ -I /usr/include/eigen3/ -lboost_unit_test_framework -lboost_filesystem -lboost_system -lboost_serialization -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world test.cpp`

The output error message is:
```
Eigen version: 3.2.92
a.out: /usr/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorStorage.h:108: void Eigen::TensorStorage<T, Eigen::DSizes<IndexType, NumIndices_>, Options_>::resize(Eigen::TensorStorage<T, Eigen::DSizes<IndexType, NumIndices_>, Options_>::Index, Eigen::array<IndexType, NumIndices_>&) [with T = double; int Options_ = 0; IndexType = long int; int NumIndices_ = 3; Eigen::TensorStorage<T, Eigen::DSizes<IndexType, NumIndices_>, Options_>::Index = long int; Eigen::array<IndexType, NumIndices_> = std::array<long int, 3ul>]: Assertion `size >= 1' failed.
Aborted
```

Thank you very much.

Leph, 2019-11-28:
As @wxmerkt pointed out, the data shoudn't be initialized as the model is empty.

closed 2019-11-28 by Leph

Leph, 2019-11-28:
It turns out my problem was related to this issue concerning the constructor of the Data structure: #641:
By adding Tensor objects to Data, using an empty model as default constructor is not possible anymore.

jcarpent, 2019-11-28:
A fix was proposed in #971