stack-of-tasks/pinocchio #972: Clone in to catkin_ws and build

micko-plz, 2019-11-28:
Hi all,
Is it possible to build pinocchio directly in the catkin workspace using catkin build, whilst also building my own catkin packages which depend on pinocchio? I'm currently having some troubles with it and what seem like suitable edits to the CMakeLists.txt file of my own ros/catkin packages not working out. Pinocchio seems to build correctly, but my own package fails with a CMake error and output:

```
  By not providing "Findpinocchio.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "pinocchio", but CMake did not find one.

  Could not find a package configuration file provided by "pinocchio" with
  any of the following names:

    pinocchioConfig.cmake
    pinocchio-config.cmake

  Add the installation prefix of "pinocchio" to CMAKE_PREFIX_PATH or set
  "pinocchio_DIR" to a directory containing one of the above files.  If
  "pinocchio" provides a separate development package or SDK, be sure it has
  been installed.

```
Thank you in advance for your time,
Michael

jcarpent, 2019-11-28:
Dear @micko-plz, 
I think the best candidate to answer your issue is @wxmerkt. Hope he will have time for it.
Best,

Justin

wxmerkt, 2019-11-28:
HI @micko-plz,
Yes, you can build pinocchio in a catkin workspace and have packages depend on it - we do this quite a lot internally. How are you trying to include Pinocchio?

Note: This is a 3rd party package built with catkin, i.e., you _cannot_ do `find_package(catkin REQUIRED COMPONENTS pinocchio)`, but you can use PkgConfig (CMake should be possible soon, but I havent tested it yet):

```cmake
find_package(PkgConfig REQUIRED)
pkg_check_modules(pinocchio REQUIRED pinocchio)
```

and then continue to use the defined variables such as `pinocchio_INCLUDE_DIRS`, `pinocchio_CFLAG_OTHERS`, etc.

An example is here: https://github.com/ipab-slmc/exotica/pull/675/files

Any further questions, please let me know!

Cheers,
Wolfgang

micko-plz, 2019-11-28:
That did the trick, wonderful. I was previously attempting the include using :
`find_package (pinocchio REQUIRED)
`
But all sorted now, thank you @jcarpent @wxmerkt for the quick response.
All the best,
Michael

closed 2019-11-28 by micko-plz