stack-of-tasks/pinocchio #973: Urdfdom shared pointer

micko-plz, 2019-11-28:
Hi again,

On integrating the changes kindly proposed in  #972 I have since come upon the issue mentioned in (https://github.com/stack-of-tasks/pinocchio/issues/654#issue-404991645), ie. the conflicting declarations between urdf_model/types.h which I guess ships with ros-melodic-desktop-full (I didn't install anything else relating to urdf's) and your own file parsers/urdf/types.hpp

The error message is 
```
/usr/local/include/pinocchio/parsers/urdf/types.hpp:31:3: error: conflicting declaration ‘typedef class boost::shared_ptr<urdf::Box> urdf::BoxSharedPtr’
   PINOCCHIO_URDF_TYPEDEF_CLASS_POINTER(Box);

```
as you're probably aware.

This type of error only came into play once I had my own package depend on pinocchio via pkg-config rather than what I previously had which was an installation from source via the instructions here: https://stack-of-tasks.github.io/pinocchio/download.html followed by a cloning of the pinocchio_catkin repository directly into my workspace.

Software versions are as follows:
Description:	Ubuntu 18.04.3 LTS
Codename:	bionic
Rosversion:     melodic

Could you let me know @wxmerkt if you have found a solution to this problem?

Thank you again,
Kindest regards,
Michael.

%%%
I'm using pinocchio to calculate the CMM, forward dynamics and their respective derivatives, for a non-linear OC problem for a humanoid-like robot. The library has been a great help so far. Thanks a million for the effort!
%%%

jcarpent, 2019-11-28:
Dear @micko-plz,
I think the issue is coming from the fact that you are not using the flags provided by the pkg-config pinocchio.pc file to compile your library which depends on Pinocchio.
For instance when writing in your terminal something like:
```
pkg-config --cflags pinocchio
```
you will obtain something like:
```
-DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM -DPINOCCHIO_WITH_HPP_FCL -DHPP_FCL_HAVE_OCTOMAP -DFCL_HAVE_OCTOMAP -DOCTOMAP_MAJOR_VERSION=1 -DOCTOMAP_MINOR_VERSION=9 -DOCTOMAP_PATCH_VERSION=0 -I/local/install/include 
```
that I've truncated.
You should use the PINOCCHIO_CFLAGS_OTHER to use these flags when compiling your own library.
For instance:
```
add_library(my_lib SHARED toto.cpp)
target_compile_options(my_lib PUBLIC ${PINOCCHIO_CFLAGS_OTHER})
```

PS: Thanks a lot for thanking the Pinocchio's development team. We really appreciate.

micko-plz, 2019-11-29:
That worked wonderfully. Thank you @jcarpent

closed 2019-11-29 by micko-plz

a-z-e-r-i-l-a, 2020-04-26:
I also get the same error, but couldn't follow the comments here about a possible solution.
I get the following output with pkg-config:
```
pkg-config --cflags pinocchio
-DPINOCCHIO_WITH_URDFDOM -DBOOST_MPL_LIMIT_LIST_SIZE=30 -I/opt/ros/melodic/include -I/usr/include/eigen3
```

micko-plz, 2020-04-27:
Hi @a-z-e-r-i-l-a , I imagine that since this issue is closed the authors won't see your question. It can be tricky to set up the CMake files at the start as it seems like just adding magic words until it works.  I'll attach the CMakeLists.txt file for my own package that uses pinocchio. I've removed most of the information unimportant to your problem, but just look for any lines containing "pinocchio" and see if you're missing them in your file. Best of luck!

```
cmake_minimum_required(VERSION x.x)
project(myproject)

# Generate compile_commands.json for clang tools
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

find_package(catkin REQUIRED COMPONENTS
        roslib
        rospy
        cmake_modules
        urdf
        kdl_parser
        robot_state_publisher
        )

find_package(Boost REQUIRED COMPONENTS
        system
        filesystem
        )

find_package(PkgConfig REQUIRED)
pkg_check_modules(pinocchio REQUIRED pinocchio)
find_package(Eigen3 3.3 REQUIRED NO_MODULE)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -fopenmp -std=c++11 -Wfatal-errors -Wl,--no-as-needed")

###################################
## catkin specific configuration ##
###################################

catkin_python_setup()

catkin_package(
        INCLUDE_DIRS
        include
        ${EIGEN3_INCLUDE_DIRS}
        LIBRARIES
        myproject
        CATKIN_DEPENDS
        urdf
        kdl_parser
        robot_state_publisher
        DEPENDS
        pinocchio
        Boost
)

###########
## Build ##
###########

include_directories(
        include
	SYSTEM
        ${pinocchio_INCLUDE_DIRS}
        ${EIGEN3_INCLUDE_DIRS}
        ${Boost_INCLUDE_DIRS}
        ${catkin_INCLUDE_DIRS}
)
link_directories(${pinocchio_LIBRARY_DIRS})

add_library(myproject
        src/file1.cpp)
add_dependencies(myproject
        ${catkin_EXPORTED_TARGETS}
        )
target_link_libraries(myproject
        ${catkin_LIBRARIES}
        ${Boost_LIBRARIES}
        dl
        )
target_compile_options(myproject PUBLIC ${pinocchio_CFLAGS_OTHER})

add_executable(node1
        src/node1.cpp
        )
add_dependencies(node1
        ${${PROJECT_NAME}_EXPORTED_TARGETS}
        ${catkin_EXPORTED_TARGETS}
        )
target_link_libraries(node1
        ${PROJECT_NAME}
        ${catkin_LIBRARIES}
        ${Boost_LIBRARIES}
        )
target_compile_options(node1 PUBLIC ${pinocchio_CFLAGS_OTHER})

#############
## Install ##
#############

install(TARGETS myproject
        ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
        )

install(DIRECTORY include/${PROJECT_NAME}/
        DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION})

install(TARGETS node1
        RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION})

install(DIRECTORY launch
        DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}
        )
```
Ps: There's a nice blog [here](https://www.siliceum.com/en/blog/post/cmake_01_cmake-basics) currently running through some cmake tutorials, particularly the CMakeLists.txt file. Not really relevant to your problem here but they encouraged me to take the time to finally figure out whats going on.