stack-of-tasks/pinocchio #975: Support data pickling

rstrudel, 2019-12-03:
Hello,

First, thanks for adding the support of model pickling I mentionned in (#957)[https://github.com/stack-of-tasks/pinocchio/issues/957] ! I noticed that the same error occurs when trying to pickle data objects. If it is possible, it would be nice to have both model and data instances picklable.

Here is a short snippet which produces an error:
`
import pickle as pkl
import pinocchio as pin

model = pin.Model()
data = model.createData()
pkl.dump(data, open('data.pkl', 'wb'))
`

Cheers,

Robin

jcarpent, 2019-12-03:
Do you really need the pickling of data? In which context?

rstrudel, 2019-12-03:
For multiprocessing purposes. If you want to compute a function which output a Data instance (using joblib or dask for example). It will output a pickling error.

Along the same lines you cannot pass a RobotWrapper instance to a function you want to parallelize because pickling is required. 

Another example is when you want to snapshot a running experiment. Pickling the current state of the robot allows you to check and debug later on.

jcarpent, 2019-12-22:
Solved by #998

closed 2019-12-22 by jcarpent