stack-of-tasks/pinocchio #976: buildReducedModel ... for geometries

nmansard, 2019-12-03:
Thanks for the nice buildReducedModel algorithm. I would need to extend the algorithm to geometry model (collision, visual). Is anybody working on this issue?

nmansard, 2019-12-03:
Here is a python version of the algo we might want to add in Pinocchio
```python
def reduceGeomModel(model,data,modelReduced,dataReduced,geomModel,geomData,q=None,qReduced=None):
    '''
    Given a kinematic model and its reduced version, a geometry model corresponding
    to the unreduced model, the corresponding data and a reference configuration (in
    both unreduced and reduced version), modify the geometry model to make it corresponds
    to the reduced kinematic tree.
    '''
    if q is not None:
        pio.forwardKinematics(model,data,q)
        pio.updateGeometryPlacements(model,data,geomModel,geomData)
    if qReduced is not None:
        pio.forwardKinematics(modelReduced,dataReduced,qReduced)

    for igeom,geom in enumerate(geomModel.geometryObjects):
        jname = model.names[geom.parentJoint]
        if(modelReduced.existJointName(jname)):
            # Case 1: the parent joint just change ID, not name
            geom.parentJoint = modelReduced.getJointId(jname)
        else:
            # Case 2: Geom is now orphan'
            # Get joint ID from the parent frame
            frame = modelReduced.frames[modelReduced.getFrameId(model.frames[geom.parentFrame].name)]
            geom.parentJoint = frame.parent
            # Change geom placement wrt new joint
            oMg = geomData.oMg[igeom]
            oMj = dataReduced.oMi[geom.parentJoint]
            jMg = oMj.inverse()*oMg
            geom.placement = jMg

```

jcarpent, 2019-12-23:
Solved by #1002.

closed 2019-12-23 by jcarpent