stack-of-tasks/pinocchio #977: Python bindings for the Lie groups

gabrielebndn, 2019-12-04:
Lie groups are missing Python bindings. So, I believe they should be implemented.
Side note: is there a way to automatically get the amount of C++ which has Python bindings, and which parts are missing them? Something like coverage, run on the code implementing the bindings?

jcarpent, 2019-12-04:
I won't have time to do the bindings, so @gabrielebndn I let you manage it. I'm not sure that these expositions are really necessary at first glance.
For the coverage, it seems to be hard, as Pinocchio is pure template ;)

gabrielebndn, 2019-12-04:
I do not have much time for that either, I was just pointing it out as something that needs to be done.
Indeed, I can just apply the straightforward formulas, I do not strictly need them right now.
But if we think that the Lie groups are useful as independent entities (as done in [this Pinocchio example](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_a-features_e-lie.html)), then I think they should be available in Python two.

Concerning the coverage: I thought that we could get coverage from the executables which are made to actually expose the files, like [bindings/python/module.cpp](https://github.com/stack-of-tasks/pinocchio/blob/6fa1c7d5502629ee126f84f1a05471815fba30f4/bindings/python/module.cpp). Do you think it is doable? Probably not, since the functions are never actually executed in those files.
But I wonder whether there is a way to obtain coverage of the C++ code from the Python unit tests

jcarpent, 2019-12-04:
So, I think you should move this topic on the board dedicated to Python bindings. I would prefer to let the issues empty (even it there is a lot of noise in the current version).

Can you do it?

nim65s, 2019-12-04:
>  But I wonder whether there is a way to obtain coverage of the C++ code from the Python unit tests

There is. It was done before. Putting this back is one of the oldest things in my todolist.

gabrielebndn, 2019-12-04:
Ok, I'll move it to the Python bindings board

closed 2019-12-04 by gabrielebndn