stack-of-tasks/pinocchio #978: Meshcat refuses to connect

gabrielebndn, 2019-12-04:
When I run https://github.com/stack-of-tasks/pinocchio/blob/devel/examples/meshcat-viewer.py
the browser opens with a window saying
```
This site can’t be reached
127.0.0.1 refused to connect.
```
I do not understand why, it was not doing it before
@wxmerkt, can you help me?
I am on Ubuntu 16.04, Chrome, python2. It is doing the same on my Macbook

gabrielebndn, 2019-12-04:
Also, the script prints
```
[11755:11755:1204/120653.296808:ERROR:viz_main_impl.cc(176)] Exiting GPU process due to errors during initialization
Fontconfig error: Cannot load default config file
[11929:1:1204/120659.640962:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request did not receive a response.
[12064:1:1204/120732.251611:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request did not receive a response.
```

jcarpent, 2019-12-04:
Have you tried:
```
pip install meshcat --upgrade
```

gabrielebndn, 2019-12-04:
Didn't work. Plus, on my Mac, it was a clean install (but my Mac is old, so probably not a good reference)

wxmerkt, 2019-12-04:
Sorry, @gabrielebndn I don't have any advice right now on how to resolve it. I won't have time before next week to investigate.

jcarpent, 2019-12-04:
@gabrielebndn I think this issue is not related to Pinocchio and should be open inside the meshcat project. I will close this issue then.

PS: on my side, everything is working well with the last version of meshcat. Try to restart your computer if needed or see with other colleagues' computer.

closed 2019-12-04 by jcarpent

gabrielebndn, 2019-12-04:
Of course I tried, but it wasn't working yesterday on my Ubuntu, it is not working today either, and it is not working with a clean reinstall on my Mac after. No idea what's going on. I agree it is an issue for meshcat. Thank you for checking on your computer