stack-of-tasks/pinocchio #980: Calculate moment of inertia around the CoM

ddliugit, 2019-12-08:
Is there a function to calculate the moment of inertia around the CoM in Pinocchio?

jcarpent, 2019-12-08:
Yes, sure. This is called `ccrba` and the desired quantity is available through `data.Ig`, the naming being related to the paper of Orin and Goswami.
```
pin.ccrba(model,data,q,v)
data.Ig
```

jcarpent, 2019-12-08:
@ddliugit It would be nice if you can write a simple example of computing and using such quantities. This small example will be helpful for other users. Thanks in advance for your contribution if you are willing to.

gabrielebndn, 2019-12-08:
I've always found that algorithm kind of funny. If all you want is Ig, then v is not needed... Instead, v is there so to compute hg from Ig... There should be st least one version of ccrba without v, I think

jcarpent, 2019-12-08:
@gabrielebndn `Ig` comes from free from many algorithms. It is not a quantity that you have to compute separately from other quantities in general ;)
Having one algorithm for each quantity contained in Data will increase a lot the current API, without any gain in performances.

jcarpent, 2019-12-10:
@ddliugit Any feedback or comments?

ddliugit, 2019-12-10:
Sorry for late reply.

I try that method, it works.

I will try to implement that, but it may take some time.

jcarpent, 2019-12-10:
OK. @ddliugit Thanks for your feedback. Hope you will be able to implement those examples.

jcarpent, 2019-12-10:
In the meantime, I close the issue as it seems to be solved!

closed 2019-12-10 by jcarpent