stack-of-tasks/pinocchio #988: Problem with the vector size

proyan, 2019-12-10:
I'm trying this simple example with `example_robot_data`, `eigenpy`, and `pinocchio`. I'm not sure where the problem is coming from, but unfortunately, this simple code is not working:

```
from example_robot_data import loadTalos
import pinocchio

robot = loadTalos()
rmodel = robot.model
rdata = robot.data
q0 = pinocchio.randomConfiguration(rmodel)

pinocchio.forwardKinematics(rmodel, rdata, q0)
```
I get the following error:
```
ValueError: "\"The configuration vector is not of right size\""
```

When I check the size being compared inside `PINOCCHIO_CHECK_INPUT_ARGUMENT` in the code, I see that `q.size()` gives a random value, and is not equal to `model.nq`.

I'm not sure if this is an installation issue, or something with example_robot_data, or eigenpy. So hints would be welcome.
Thanks.

cmastalli, 2019-12-10:
It seems to be related to https://gepgitlab.laas.fr/gepetto/example-robot-data/issues/17. Could you try to use NumPy matrix before loading the robot?
```python
from example_robot_data import loadTalos
import pinocchio

pinocchio.switchToNumpyMatrix()
robot = loadTalos()
rmodel = robot.model
rdata = robot.data
q0 = pinocchio.randomConfiguration(rmodel)

pinocchio.forwardKinematics(rmodel, rdata, q0)
```

proyan, 2019-12-10:
Does the code sample above work for you? I just want to know whether this is an installation issue.
I don't think the issue is with example_robot_data. It is either in my installation, or in pinocchio.

proyan, 2019-12-10:
Confirmed with Carlos. This is an installation issue.

proyan, 2019-12-10:
I'm using `eigenpy` from devel, and `pinocchio` from devel. I moved `eigenpy` to `v1.6.9`. Now I don't have the problem anymore.

This issue should be transferred to https://github.com/stack-of-tasks/eigenpy

jcarpent, 2019-12-10:
Why should be transfered to eigenpy?

closed 2019-12-10 by jcarpent

proyan, 2019-12-10:
Because this seems to be arising because I was using the devel version of eigenpy. When I switch to the last release v1.6.9, it disappears.

muzi2018, 2022-02-12:
maybe :  the type of joint in your urdf isn't supported by pinocchio
pinocchio just support the following joint type:
https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_c-maths_c-joints.html
Revolute
Prismatic
cylindrical
Spherical
Translation
Free-flyer