stack-of-tasks/pinocchio #993: Expose data.dhg in python

ddliugit, 2019-12-12:
I just happen to use `computeCentrodialMomentumTimeVariation` to compute angular momentum derivative and I can find `data.hg`, but the `data.dhg` is missing in python.

The install procedure follows the website.
`https://stack-of-tasks.github.io/pinocchio/download.html`

I do find a issue which is relevant to this#760, but I didn't see you add `dhg` in the code.

jcarpent, 2019-12-12:
Done through #994. Thanks for the report.

ddliugit, 2019-12-12:
Thanks for your timely reply. 

Do I need to build from source? Or should I repeat the same procedure in this website to install from binary?

I am using Mac and linux

ddliugit, 2019-12-12:
I download the dependencies, revised the code by following your code, recompile it and install it. 

The error still appears:
```
('Simulator: Linear momentum:', [0.0, 0.0, 0.0])
('Simulator: Angular momentum w.r.t CoM:', [0.0, 0.0, 0.0])
Traceback (most recent call last):
  File "data_collecting_biped.py", line 341, in <module>
    print('Simulator: Linear momentum time derivative:', vec2list(bipd.pinocchio_robot.data.dhg.linear))
AttributeError: 'Data' object has no attribute 'dhg'
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

```

ddliugit, 2019-12-12:
Did I miss something? I only add the following in the `data.hpp`
```
        .ADD_DATA_PROPERTY_READONLY(Force,dhg,
                                    "Centroidal momentum time derivative(expressed in the frame centered at the CoM and aligned with world frame).")
```

jcarpent, 2019-12-13:
A simple brew reinstall pinocchio —head should work.

closed 2019-12-13 by jcarpent

ddliugit, 2019-12-13:
For linux, without brew, how to do that

gabrielebndn, 2019-12-13:
@ddliugit, since the change is not released, the only way to have it now it to build it from source on the branch devel. You can find instructions [here](https://stack-of-tasks.github.io/pinocchio/download.html)

ddliugit, 2019-12-13:
I did that, but it doesn’t work. Should I use different cmake flag

gabrielebndn, 2019-12-13:
I do not think it has anything to do with the CMake flags.
1. If you were using APT binaries before, make sure to remove them:
```
sudo apt remove robotpkg-py27-pinocchio
```
2. If you have made changes to your repo, clean it:
```
git fetch --all
git reset --hard origin/devel
rm build*
```
(assuming the main Pinocchio remote is called origin).
This will completely clean your repo of any change you have made to the repo. Then, you can proceed with a new installation.
While installing, pay attention to the installation path: `sudo apt-get` installs in `/opt/openrobots`. You can control where you install when doing
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/your/installation/path
```
You can install in `/opt/openrobots`, but this is not recommended. Choose another path. Then, you will have to configure your environment variables as
```
export ROBOTPKG_BASE=/your/installation/path
export PATH=$ROBOTPKG_BASE/bin:$ROBOTPKG_BASE/sbin:$PATH
export LD_LIBRARY_PATH=$ROBOTPKG_BASE/lib:$ROBOTPKG_BASE/lib/plugin:$ROBOTPKG_BASE/lib64:$LD_LIBRARY_PATH
export PYTHONPATH=$ROBOTPKG_BASE/lib/python2.7/site-packages:$ROBOTPKG_BASE/lib/python2.7/dist-packages:$PYTHONPATH
export PKG_CONFIG_PATH=$ROBOTPKG_BASE/lib/pkgconfig/:$PKG_CONFIG_PATH
export ROS_PACKAGE_PATH=$ROBOTPKG_BASE/share:$ROBOTPKG_BASE/stacks:$ROS_PACKAGE_PATH
export CMAKE_PREFIX_PATH=$ROBOTPKG_BASE:$CMAKE_PREFIX_PATH
```

ddliugit, 2019-12-13:
Thanks. 
I think it is caused by binary problem. First I followed robotpkg installation. I will do remove thing.