stack-of-tasks/pinocchio #999: BUILD_TESTING vs BUILD_UNIT_TESTS

gabrielebndn, 2019-12-18:
Module CTest automatically creates a `BUILD_TESTING` option that selects whether to enable testing support, ON by default, see [here](https://cmake.org/cmake/help/latest/module/CTest.html).
In Pinocchio, this is included by the cmake submodule by [post-project.cmake](https://github.com/jrl-umi3218/jrl-cmakemodules/blob/54ece258eed16da94c7f10979588fcb47b744eb8/post-project.cmake#L44). This variable is then employed consistently by macro `ADD_UNIT_TEST`, defined in the cmake submodule in [test.cmake](https://github.com/jrl-umi3218/jrl-cmakemodules/blob/54ece258eed16da94c7f10979588fcb47b744eb8/test.cmake#L72), to control whether unit tests are built or not.

However, if one is building Pinocchio from source, the variable to control the building of unit tests is `BUILD_UNIT_TESTS`. `BUILD_TESTING` is then brutally set to `BUILD_UNIT_TESTS` in [the main CMakeLists.txt](https://github.com/stack-of-tasks/pinocchio/blob/474d0706d44286476d8956d43c41e6a7bcd2350b/CMakeLists.txt#L64). As you can see, the original value of `BUILD_TESTING` is completely ignored.

As a result, there are two variables which exist to control the compilation of the unit tests, but one of them, is ignored and overridden by the other. When doing `ccmake`, I see this variable `BUILD_TESTING` which is set to `ON` and does nothing.

I think this is confusing, especially to newcomers. The worst part is that `BUILD_TESTING` is standard CMake, therefore it is naturally the variable of choice for those who who first delve into Pinocchio with some CMake experience; but setting this variable manually has no effect at all.

I think the main variable to control the unit tests should be `BUILD_TESTING` (defaulting to OFF). `BUILD_UNIT_TESTS` should be deprecated and later removed altogether. What do you think?

jcarpent, 2019-12-18:
I do agree. We can indeed remove it.

jcarpent, 2019-12-19:
Solved by #1000.

PR #1000 (2019-12-19): Deprecate BUILD_UNIT_TESTS. Closes #999
This closes #999
commit 0708f314 (2019-12-19): Merge pull request #1000 from gabrielebndn/topic/BUILD_UNIT_TESTS

Deprecate BUILD_UNIT_TESTS. Closes #999
closed 2019-12-19 by jcarpent