stack-of-tasks/pinocchio #1005: Force HPP-FCL bindings dependency

gabrielebndn, 2020-01-07:
Currently,  Pinocchio Python bindings can be built:
1. Without FCL
1. With FCL, but without FCL bindings
1. With FCL, including the bindings

I believe the second option is unreasonable and only adds complexity. If Python bindings are built with HPP-FCL, then HPP-FCL bindings should be enforced. Ideally, we should add something like this to the [CMakeLists.txt](https://github.com/stack-of-tasks/pinocchio/blob/e038c7bf283b1df56a35014455e0e2d6f36e03ac/CMakeLists.txt):

```
IF(BUILD_PYTHON_INTERFACE AND BUILD_WITH_COLLISION_SUPPORT)
  ADD_REQUIRED_DEPENDENCY("hpp-fcl-python")
ENDIF(BUILD_PYTHON_INTERFACE AND BUILD_WITH_COLLISION_SUPPORT)
```

so we need a way to detect hpp-fcl Python bindings via cmake.

What do you think?

jcarpent, 2020-01-07:
We cannot check FCL bindings with pkg-config. I mean this is not a standard way of doing it. In addition, we are not sure yet to get the latest FCL version on all the platforms.

@nim65s Is hpp-fcl latest release available on robotpkg?

To do it correctly, I would suggest using directly the packaging of CMake, which allows exporting additional targets, such as the Python bindings.

gabrielebndn, 2020-01-07:
> In addition, we are not sure yet to get the latest FCL version on all the platforms.

For this, it will be enough to do a new HPP-FCL release and then increase the minimum version required by Pinocchio. This way everybody using HPP-FCL will be forced to upgrade (I do not see why anybody should want to keep old versions).

> To do it correctly, I would suggest using directly the packaging of CMake, which allows exporting additional targets, such as the Python bindings.

Yes, I think this may be the correct way.

jcarpent, 2020-01-08:
Changing the minimal version number is indeed in the pipe. As I mentioned previously,I’m waiting the update from robot from robotpkg first. I will close the issue and I will take care of all it soon.

closed 2020-01-08 by jcarpent

nim65s, 2020-01-08:
On robotpkg, hpp-fcl is on v1.2.1. The current latest is 1.2.2. I'll update it.

nim65s, 2020-01-10:
@jcarpent  1.2.2 is available

jcarpent, 2020-01-13:
@nim65s Thanks a lot!