stack-of-tasks/pinocchio #1007: Documentation is not up to date

gabrielebndn, 2020-01-10:
The documentation https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/ is not up to date.
The latest release is 2.2.3 (the latest apt release is 2.2.2 instead), but the online doc appears to be 2.2.1.
I know this can be fixed by manually copying the doc on `/net/pongo`, but I think there should be an automatic solution. @nim65s?

jcarpent, 2020-01-10:
This is already done normally by the internal tools of LAAS, but I remember that @nim65s told be that there are some trubles. I think you should discuss it in private with him.

gabrielebndn, 2020-01-10:
Yes, I remember that. I just wanted to point to the fact that the problem is still there. I'll close the issue

closed 2020-01-10 by gabrielebndn