stack-of-tasks/pinocchio #1011: checkData fails on empty model

jmirabel, 2020-01-13:
The following code returns 1 (checkData returns false):
```c++
#include "pinocchio/multibody/data.hpp"
#include "pinocchio/multibody/model.hpp"
#include "pinocchio/algorithm/check.hpp"

using namespace pinocchio;
int main()
{
    Model empty_model;
    Data empty_data(empty_model);
    if (!pinocchio::checkData(empty_model, empty_data)) return 1;
    return 0;
}
```

By the way, when try to compile the unittest on v2.2.3, I get the following error
```
Scanning dependencies of target data-test
[ 66%] Building CXX object unittest/CMakeFiles/data-test.dir/data.cpp.o
In file included from pinocchio/build/include/pinocchio/multibody/data.hpp:9:0,
                 from pinocchio/unittest/data.cpp:5:
pinocchio/build/include/pinocchio/math/tensor.hpp: In member function ‘void pinocchio::Tensor<Scalar_, NumIndices_, Options_, IndexType>::resize(const Eigen::array<IndexType, NumIndices>&)’:
pinocchio/build/include/pinocchio/math/tensor.hpp:252:68: error: there are no arguments to ‘coeffRef’ that depend on a template parameter, so a declaration of ‘coeffRef’ must be available [-fpermissive]
         if(size_changed) EIGEN_INITIALIZE_COEFFS_IF_THAT_OPTION_IS_ENABLED
```

jcarpent, 2020-01-13:
Which version of Eigen are you using?

jmirabel, 2020-01-13:
3.2.92

jcarpent, 2020-01-13:
Thanks. I have fixed the bug related to the Tensor class.
But, unfortunately, I was not able to reproduce the bug for the checking of the data. I will wait for Travis and co to provide some feedback on it.

jmirabel, 2020-01-13:
with gdb, this faulty line seems to be:
https://github.com/stack-of-tasks/pinocchio/blob/e038c7bf283b1df56a35014455e0e2d6f36e03ac/src/algorithm/check.hxx#L124

jcarpent, 2020-01-13:
I see. This is because there is a bug in Eigen 3.2.92 with the Tensor that cannot be initialized with zero-dimensions. I will provide a fix for it. Thanks a lot for this report!

jcarpent, 2020-01-13:
I've fixed the bug on the current open PR #1012. Hope it will work now.

jcarpent, 2020-01-14:
Solved by #1012.

closed 2020-01-14 by jcarpent