stack-of-tasks/pinocchio #1015: Walking robot ground contact question

mklvcm, 2020-01-14:
Sorry for a newbie question.

I'm trying to use Pinocchio to drive a walking robot (both simulator and real) walking on uneven terrain.
I'm not sure what would be the rigth thing to do when foot couches the ground.

In simulator, do I need to run collision detection on each iteration to see if it stands on the ground?
If contact detected, how do I take it into account? Do I supply it as external force to RNEA? Do I change joint type to fixed?

In real robot I have touch sensors.
This makes problem easier - no collision detector is required.
However now that foot is in contact, what do I specify this fact when I do RNEA calculations?

Thank you.

jcarpent, 2020-01-15:
Dear @mklvcm,

You've several possibilities. You can do either Task Space Inverse Dynamics (instantaneous control) or Model Predictive Control (preview control).

In your case, you have to compute the correct contact forces when there are some parts of the robots that are in contact in order to drive your motion while avoiding the fall. This can be stated as QP problem where the contact forces (when there is contact) have to be inside coulomb friction cones. 
In the context of your quadruped, it is easy to detect contact, because you have spherical feet, and to know the contact normal without using any collision library.
For computing the desired contact forces you have to apply on the ground to stand, I would recommend you relying on [TSID](https://github.com/stack-of-tasks/tsid) which is derived from Pinocchio to solve the kind of problem you're focusing on. It comes with Python bindings for easy code prototyping as well as tutorials.

After computing these contact forces, you can then use the ABA algorithms with external contact forces to do the simulation if you want.

Best,

Justin

jcarpent, 2020-01-20:
@mklvcm Any news?

mklvcm, 2020-01-27:
Hi Justin,

thanks for getting back to me and asking if I have any news! So nice :)

Just to give you a context of what I'm doing, see this: https://www.youtube.com/watch?v=TiRZ2_23Esk 
I designed that from scratch and I'm now trying to do software.
I'm absolute newbie to the field, I'm a bit overwhelmed :)

Well I was looking at TSID library that you suggested.
I can see what it does but maybe it does a bit too much.
What I was looking for is simply add/removeRigidContact method, if it was available in pinocchio that would be great.

I'm also looking at bullet3 library.
If you have any other suggestions please let me know!

Thanks for your help
Misha.

jcarpent, 2020-02-19:
I will close the issue. @mklvcm Let us know if you need more help.

closed 2020-02-19 by jcarpent