stack-of-tasks/pinocchio #1016: [Bindings] container::aligned_vector<Force> in python

NoelieRamuzat, 2020-01-15:
Hi !
I would like to use the `rnea` algorithm with the external forces in python.
The binding of the method `rnea(Model, Data, Configuration q, Velocity v, Acceleration a, External forces f)` does exist.
But the external forces should be expressed as a `container::aligned_vector<Force>` vector, which is an object of pinocchio.
I have seen that there is a binding in [std-aligned-vector.hpp](https://github.com/stack-of-tasks/pinocchio/blob/master/bindings/python/utils/std-aligned-vector.hpp). But I don't know how to use it.

In python I have access to `pin.utils.pin.StdVec_Force`, should I use it ? 
My external forces are expressed as a numpy.ndarray, how can I convert it ? 

Thank you very much for your time !

jcarpent, 2020-01-15:
You should indeed use `pin.StdVec_Force` which has the same API than a list in Python.
Then, you should fill this `pin.StdVec_Force`with all the external forces for each joint expressed in the local frame of joint.

jcarpent, 2020-01-15:
One thing to do is to extend the current bindings with functions taking Python list as input and converting them to std::vector.

NoelieRamuzat, 2020-01-16:
Hi sorry for the delay, 
It is working well thank you !
I've created a `pin.Force`object for each forces I had and append them to a ` pin.StdVec_Force`.
The `rnea` method works well then with this vector.
Thus, I close the issue.

closed 2020-01-16 by NoelieRamuzat