stack-of-tasks/pinocchio #1019: Instalolation on Ubuntu 19.04

gabrielebndn, 2020-01-20:
Hallo!
I am trying to install Pinocchio via apt on Ubuntu 19.04, following the procedure outlined [here](https://stack-of-tasks.github.io/pinocchio/download.html).

Everything goes smoothly until `sudo apt-get update`. When I do it, I get the error:

```
E: The repository 'http://robotpkg.openrobots.org/packages/debian/pub eoan Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

I could dig into `apt-secure`, but I am reluctant to change security-related settings.
Any explanation about the cause of this error?
Can it be fixed at the robotpkg level so that installation would work on the fly?

stonneau, 2020-01-20:
group 20
​
buttons as well as app (2 interqctions means)​
arent using object recogbition just trays​
just forget buttons​
localisation (no gps in room)​
qpp for elderly​
turtlebot 20 kg​
clear sketch / ​
clear market for product , more research on topic​
​
​
19​
gambling no good app domain (card games, no poker) ​
is it too hard ?​
navigation around table too hard(obstacles) ? does it really need to move ?​
what sensors used ?​
no point in using card substitute, find way to use cards. Simpify design (succion cups)​
​
​
18​
thought of adjustments of axes, nice idera​
prototype relevant​
whole robot should not be only remote controlled ​
Security / safety in general must be addressed​
good teamwork in oral ​
are they going to be able to avoid objects​
whats the detailed used cases addressed ? be more specific​
Omnidirectional zheels nice idea ​
​
17​
nice name​
nice thinking about design / details​
really complicated mechanical design​
don t do it zith lego ​
consult users​
speech seems difficult​
user needs control to be specified​
​
16​
design unclear, diff between real library and mock up​
should recognize current labeling of books​
reasonable need for project which is nice​
consult real librearians​
finding lost books ? (scan shelves)​
try to think about existing sytems and possible interacing polssib.​
line following too easy, not practical​
​
15​
Actual project design and use cases not clear​
Need to talk to users to find actual use cases, both nurses and patients​
is this at home / hospital ? ​
human in the loop design (button to say people ok) good idea​
focus on one or 2 good use cases​
​
14 * needs to rethink ?​
presentation nice​
missing resaerh in the field​
usability (how do you wash it). Robotics and food seems complex​
app interface is a bad idea ​
talk to targer audience who cook a lot to find where they are losing time​
Safety is a concern here​
Cutting food really hard problem ​
​
13​
do your research on what already exists and how are you different​
design issues: clarify video system​
clarify market, renters / owners​
security of app, door lock system really important ​
nice idea otherwise, ineresting potential​
specify and handle failure cases ​
​
12​
use case is nice Experts acually disagree about whether this is a good idea.​
is the current design pratical ?​
How does it scale as a product ? ​
Interesting robotic problem​
how this verify correct operation ?​
Magnetic autonomuos version for drawing ?​
​
11​
No mechanical designs​
Is it one or 2 robots, does it need special shelf ​
look at previous sdp prject from last year​
ok to expeand, need somthg new​
how does it check correctness of behaviour​
specify navigation system​
​
10 *​
Could develop ;odule to existing teaching platform​
no mechanical design specification​
potential good thing: more software focus assistant ​
​
9:​
Motivate through research why not static system​
are they going to go over actrual mud / rough terrain ?​
find of better use case (greenhouse)​
use infrared camera to monitor ?​
what is the data required / sensors that allow it​
​
8: *​
No mechanical design presented, though challenging topic​
specify how sorting is made and what elements sorted​
look into last year projects and see how to expand​
this is a HARD project​
what is target success rate ?​
try to filter to make sure success​
​
7:​
very creative idea​
business case doubtful​
vs One robot that moves the table​
Generalize to chairs ?​
Specify planning, localisation, nav​
​
​
6:​
Should it adapt to human behaviour ? (Can person hang to it / trolley design ?)​
POssibly to carry people ?​
Interesting software topic​
​
5:​
Should use sensors​
force closure is the challenge. Might need to find clever way to address this​
design of own jars necessary ?​
How Omni will it be ?​
what is the demonstration goal​
​
4:​
Motivate better need for moving robot​
Better identify use cases: Home / outside ​
Make sure you don t remove joy humans get from this​
Look at last year s project / expand​
We insist that use case must be specified​
How to handle water in mockup​
Define how to deal with diff plants, what criteria relevant​
​
3:​
Appreciated extensive research​
why not just and app ? Motivate better vs app​
mechanical design warning​
budget issue​
control​
​
2:​
​
​

gabrielebndn, 2020-01-20:
@stonneau I think you accidentally sent me your planning :D

stonneau, 2020-01-20:
sorry :)

jcarpent, 2020-01-20:
@gabrielebndn EOAN is the name for Ubuntu 19.10, which is not yet supported by robotpkg. Please use either an LTS or 19.04.

nim65s, 2020-01-20:
19.04 is supported by robotpkg, in the sense that builds are tested on it:http://robotpkg.openrobots.org/rbulk/robotpkg/index.html
But I don't think it has binary builds:  http://robotpkg.openrobots.org/debian.html

If you need that at wondercraft, we can probably add it.

gabrielebndn, 2020-01-20:
> @gabrielebndn EOAN is the name for Ubuntu 19.10, which is not yet supported by robotpkg. Please use either an LTS or 19.04.

@jcarpent oh! Hadn't seen that! Thanks!

gabrielebndn, 2020-01-20:
> 19.04 is supported by robotpkg, in the sense that builds are tested on it:http://robotpkg.openrobots.org/rbulk/robotpkg/index.html
> But I don't think it has binary builds: http://robotpkg.openrobots.org/debian.html
> 
> If you need that at wondercraft, we can probably add it.
> 
> edit: the list Debian (and derivatives) versions with binary packages is also here: http://robotpkg.openrobots.org/packages/debian/pub/dists/ so yes, my first link has out of date content, and 19.04 (disco) is available.

@nim65s as @jcarpent made me notice, I am actually on 19.10. Any plans to release Pinocchio on 19.10?

nim65s, 2020-01-20:
The VM should be upgraded to 19.10, yes, but nobody ever asked for it, so there is no priority on this at all. 

So if this is going to be really important for you, we can try to speed up on this point.

But if this is only for a corner case on one development laptop that can be easily workarounded, we could as well wait for 20.04 LTS :)

gabrielebndn, 2020-01-20:
@nim65s well, it's on my desktop computer, so it would be handy for me in this moment, but I don't know how many people in Wandercraft use it, I will try to update you soon.
As for myself, it is not that important, and I can easily resort to a source build anyway

gabrielebndn, 2020-01-20:
I'll close the issue since the problem is now clear. Thanks for the clarifications!

closed 2020-01-20 by gabrielebndn