stack-of-tasks/pinocchio #1021: Compiling for Python 3

gabrielebndn, 2020-01-21:
How do I compile Pinocchio for Python 3?

jcarpent, 2020-01-22:
You should search on the web a bit for that normally. You should easily find a solution.
A solution is:
```
cmake .. -DPYTHON_EXECUTABLE=${which python3) -DPYTHON_LIBRARY=path_to_your_python_lib
```

gabrielebndn, 2020-01-22:
Well, I saw many Python-related options and these things tend to be tricky, with potential for a huge mess. I will try your suggestion, thanks. I suggest to add a single unambiguous option that takes care of this to facilitate the process, especially a Python 2 is being phased out. Also, Python 3 should probably become the default quite soon

gabrielebndn, 2020-01-22:
Anyhow, I had searched in the doc, but found nothing. Maybe some information about it could help too

jmirabel, 2020-01-22:
I think `cmake .. -DPYTHON_EXECUTABLE=${which python3)` is sufficient, isn't it?

jcarpent, 2020-01-22:
I've tried on mac, and it was not (because if you have several Python 3 on your OS, it won't use the right one).

nim65s, 2020-01-22:
`-DPYTHON_EXECUTABLE=$(which python3)` always works for me, except on a dirty build directory where python2 has already be found. In this case… Clearing the build directory is enough :)

jcarpent, 2020-01-22:
> I suggest to add a single unambiguous option that takes care of this to facilitate the process, especially a Python 2 is being phased out. Also, Python 3 should probably become the default quite soon

If it had been easy to do it, for sure we would have done it for sure. The problem is that on your OS you may have several Python 3.x installed, and then, most of the scripts coming with cmake failed.

nim65s, 2020-01-22:
The default option is decided at the OS level: if the main `python` interpreter of the system (usually `/usr/bin/python`) is python 3, you will have Python 3. On Arch, it's already Python 3 by default, and hopefully, in 20.04 it will also be the case. So we don't need to take any action here.

This would be different if we decided to drop python 2 support ; in this case it would make sense to enforce python 3 at the CMake level. But it would not be a nice move in Pinocchio for now.

jcarpent, 2020-01-22:
The question is also when you have Python3.6 and Python3.7 installed, often, I have seen the script from cmake failling. But this is only my personal experience with cmake.

gabrielebndn, 2020-01-22:
I confirm that simply `-DPYTHON_EXECUTABLE=$(which python3)` works for all packages I tried (eigenpy, hppfcl and pinocchio) on Ubuntu 19.10. The system was basically clean, with no previous Python2 installations.
I'll let you know if I try it on my Mac (which is also quite dirty).
Thanks a lot for your help!
I'll close the issue.

closed 2020-01-22 by gabrielebndn