stack-of-tasks/pinocchio #1026: Switching to array in Pinocchio

gabrielebndn, 2020-01-24:
Hi there!
Python2 has reached its end of life, and `numpy.matrix` is now officially deprecated.
As such, there is now a strong push to switch to `matrix`. In particular, we are considering this in Wandercraft.
In general, Pinocchio methods handle this through eigenpy, with its methods `switchToNumpyMatrix` and `switchToNumpyArray`.
However, there is a small subset of Pinocchio methods which are purely written in Python, such as [utils.py](https://github.com/stack-of-tasks/pinocchio/blob/b9d99528d2bff742d406179e8bf2cc9999f405cd/bindings/python/pinocchio/utils.py
) and [rpy.py](https://github.com/stack-of-tasks/pinocchio/blob/b9d99528d2bff742d406179e8bf2cc9999f405cd/bindings/python/pinocchio/rpy.py).
These methods were written when `matrix` was the standard, and they generally don't support `array` well: either they assume `matrix` as input, or they return `matrix`.
This should be fixed. I opened this issue to highlight the problem and to discuss about how to do it exactly. My suggestion to ensure a smooth transition:
- Modify the methods so that they work for both `matrix` and `array` inputs, analogously to what done in https://github.com/stack-of-tasks/pinocchio/pull/951, using `dot` instead of `*`
- Concerning the return type: check `eigenpy.getNumpyType()` and do the same as eigenpy

Alternatively: drop support for matrix (but I feel this is premature).

jcarpent, 2020-01-24:
Yes, you can proceed to the support of both I think, knowing that some functions are already in C++, it would be worthy to expose them instead of coding them in Python ;)

jcarpent, 2020-01-29:
@gabrielebndn What is your timeline?

gabrielebndn, 2020-01-29:
I don't have a set timeline but I am working on it. I will make a PR for the RPY module

jcarpent, 2020-01-29:
Solved by #1042.

gabrielebndn, 2020-01-29:
I am sorry, but #1042 di not solve this issue. There are still things to do in the other modules

jcarpent, 2020-02-02:
#1050 has solved many point of this issue. We can now consider this issue as solved.

closed 2020-02-02 by jcarpent

gabrielebndn, 2020-02-03:
@jcarpent if you are convinced that the transition is completed, could you publish a new release, at leat a source release? It would help us switching to `numpy.array` here at Wandercraft

gabrielebndn, 2020-02-03:
I recommend increasing at least the minor version: `2.3.0`

wxmerkt, 2020-02-03:
Since this is change to `np.array` is already an API breaking change, should we also include #1025?

jmirabel, 2020-02-03:
This isn't an API breaking change, isn't it ? At least, it shouldn't be.

gabrielebndn, 2020-02-03:
> Since this change to `np.array` is already an API breaking change, should we also include #1025?

Well, it's not exactly the same thing. The API change is not in Pinocchio, but in eigenpy, which was increased in major version switching to 2.0.0. Actually, the whole adaptation work I did was to ensure that the current API would _still_ work even with array

gabrielebndn, 2020-02-03:
Code will be broken for those who are using matrix and haven't put `switchToNumpyMatrix()` in their code. But again, this is something that concerns eigenpy, rather than Pinocchio, and we have been issuing warnings for a long time about this change

jcarpent, 2020-02-03:
> Since this change to `np.array` is already an API breaking change, should we also include #1025?

For this, we can add a compilation flag 'PINOCCHIO_ENABLE_3x_FEATURES` if you want.

jcarpent, 2020-02-03:
> Code will be broken for those who are using matrix and haven't put `switchToNumpyMatrix()` in their code.

Yes, but they can still fix it by just calling `switchToNumpyMatrix`, so no worry about it.

jcarpent, 2020-02-03:
Before making the release, I would like to fullfil this issue #914. Someone is on it now?

gabrielebndn, 2020-02-11:
@jcarpent, I would really appreciate it if you could publish a new release.
Can #914 wait for the next one?

jcarpent, 2020-02-14:
I would prefer to have it fixed, it is a matter of version numbering and also, it is important for all the partners. I would really appreciate if someone other than me can handle it.

gabrielebndn, 2020-02-14:
Impossible for me. I think it would not be really time-consuming if I knew exactly how to do it, but the truth is that I don't and that I have zero time :grimacing: