stack-of-tasks/pinocchio #1044: Linear Interpolation between two SE3s

ddliugit, 2020-01-30:
I am working on a problem which needs to insert interpolation poses between initial SE3 and final SE3 with respect to time knots.

I am wondering if there is a library to do that in Pinocchio?

jcarpent, 2020-01-30:
You can use directly the following example: https://github.com/stack-of-tasks/pinocchio/blob/master/examples/interpolation-SE3.cpp

ddliugit, 2020-01-30:
I am using python library. How can I expose this API to python

jmirabel, 2020-01-30:
As far as I know, it isn't bound.

ddliugit, 2020-01-30:
how to bind that API  with python?

gabrielebndn, 2020-01-30:
I confirm it is not bound. There is a specific issue on it, but we closed it because we did not plan to work on it https://github.com/stack-of-tasks/pinocchio/issues/977, it can be found in [the Python bindings project](https://github.com/stack-of-tasks/pinocchio/projects/3).

 @ddliugit if you want to make the bindings you can take inspiration from the existing ones (https://github.com/stack-of-tasks/pinocchio/tree/master/bindings/python). We would all be very grateful if you made this addition to Pinocchio! But I warn you it requires a considerable level of expertise and it is no easy task for a newcomer

ddliugit, 2020-01-30:
Thanks. 

Converting to python is difficult and I will try to do that if I am available.

closed 2020-01-30 by ddliugit

jmirabel, 2020-01-30:
Anyway, @ddliugit for your specific problem, you can workaround the issue by using something like

    M = a * exp6( u * log6(a.inverse()*b))

where a and b are your SE3 objects, and u is in [0,1].

ddliugit, 2020-01-30:
Thanks!

That's a good idea. I will try that!

rancheng, 2022-02-17:
I wrote a script to do the interpolation: 

[https://github.com/rancheng/SE3_Pose_Interp](https://github.com/rancheng/SE3_Pose_Interp)

hope it can help!

jmirabel, 2022-02-17:
I couldn't find the info of what kind of interpolation you do. Slerp between two consecutive transforms ?

ddliu365, 2022-02-17:
@rancheng thanks for providing this script. I actually used another library ([ndcurves](https://github.com/loco-3d/ndcurves)) for the final implementation. Just post here for your reference.