stack-of-tasks/pinocchio #1047: Does Pinocchio provide an analytical expression for the mass matrix ?

vpadois, 2020-01-31:
Hello,

I am not sure that the following feature comes with Pinocchio : analytical expression for the mass matrix. Is it the case ? If yes is there a running example that I could rely one ?

Thanks in advance.

best
Vincent

jcarpent, 2020-01-31:
Hi Vincent,

Nice to meet you here.

What do you mean an analytical expression:
- the computation graph?
- the whole formula?

Best,

Justin

vpadois, 2020-01-31:
From our perspective we could potentially make use of both. I guess you rather provide the graph than the whole formula ?  Or can you do both by default ?

Thanks
Vincent

jcarpent, 2020-01-31:
Thanks to CppAD, we can do both in fact.

jcarpent, 2020-01-31:
When do you need it? 

With the current status of Pinocchio, I can only provide you a C++ example. Is it fine?

vpadois, 2020-01-31:
> When do you need it?

Whenever you can  / the sooner the better  :slightly_smiling_face: 

> With the current status of Pinocchio, I can only provide you a C++ example. Is it fine?

Sure.  We are actually planning to use it with a 7 dof manipulator.

Thanks in advance. Let us know what we can do to help.

jcarpent, 2020-02-05:
@vpadois I'm on it now, just to be sure of your needs. Do you only want to:

- evaluate the Mass Matrix according to the current configuration of the robot, in this case:
`crba(model,data,q)` will do the job
- having a function of type `computeMassMatrix(q)`, in this case code generation will do the job
- or having a Latex file with the whole formula

Pinocchio can provides the three, but I would prefer to not waisting time by providing a side feature which is not needed.
Thanks in advance for your quick reply.

vpadois, 2020-02-06:
Thx @jcarpent.

What we really need is a symbolic expression of the mass matrix, possibly in the form of a function.  I guess this corresponds to the second option you describe ? The more this expression is factorized with respect to the symbolic variable q, the better it is for us.

best
Vincent

jcarpent, 2020-02-06:
Dear @vpadois, you can find the solution of your issue in #1059.
I will add the feature for generating latex equations soon.

vpadois, 2020-02-06:
> Dear @vpadois, you can find the solution of your issue in #1059.
> I will add the feature for generating latex equations soon.

Thank you so much for your reactivity. We will work on it and give you feedback as soon as possible.

Best
Vincent

closed 2020-02-06 by vpadois

rshum19, 2020-04-17:
Dear @jcarpent,  I would be very interested in an example of generating latex equations using CppAD,  do you think you would be able to add this feature soon? 

Thank you!