stack-of-tasks/pinocchio #1052: Why is there no reverseDynamics with contacts?

mklvcm, 2020-02-02:
Maybe another ignorant question, sorry.
I can see forwardDynamics function that takes contacts as parameters.
But inverseDynamics does not.
Why is that?
And what would be a way to implement it?

jcarpent, 2020-02-03:
Dear @mklvcm,

You can have a look at [TSID](https://github.com/stack-of-tasks/tsid) repository, which is a layer on top of Pinocchio to do what you want to achieve.
Best,

Justin

jcarpent, 2020-02-03:
I will close this issue as it seems to be solved. @mklvcm Feel free to reopen it if needed.

mklvcm, 2020-02-03:
Hi Justin,

I'm afraid I'm testing your patience here and annoy the hell out of you but I can't help but ask :)
Similar question was asked long time ago here: https://github.com/stack-of-tasks/pinocchio/issues/61
Back then you said "You just need to compute the jacobian of contacts..." and apparently topic starter was happy.
Because I'm more ignorant than that guy, can I ask you to provide a sample of code, even if pseudo-code.

Thanks
Misha.

PS. No I cannot re-open issue if closed by you.

jcarpent, 2020-02-03:
A first answer was indeed provided in #61, in the case where you know the contact forces you may want to apply, for instance, in the context where the contact forces are related to the sole deformation with a spring-damper model.

Before going further into my answer, could you provide a typical setup where you want to perform inverse dynamics?

mklvcm, 2020-02-04:
Hi Justin,

My case is 4-legged robot walking on uneven terrain.
I'm trying to use inverseDynamics to calculate torques.
I understand that I need to know external forces but I don't know them.
Instead I know which out of 4 legs are in contact with the ground.
If there was an overload of inverseDynamics that takes contacts that would solve my problem.
Now I see that such overload exists for forwardDynamics and I'm wondering why not for inverseDynamics? And how hard would it be to write one? 
Any code sample would be appreciated!

Thanks,
Misha

jcarpent, 2020-02-05:
@mklvcm Sorry for the late reply, but a quick answer to your question is that:
while the forward dynamics is a well-posed problem because for any torque you apply on the joints of the robot, then you can get the joint acceleration of the system submitted to gravity, the inverse dynamics is an ill-posed problem in the sense you don't know the actuation for the free-floating joint of your robot.

That's why you have to resort on TSID, which means Task Space Inverse Dynamics, which allows you to compute one (possible) solution.

closed 2020-02-06 by mklvcm