stack-of-tasks/pinocchio #1055: bug report : CHECK_DATA for data.kinematic_hessians.dimension

Linjackffy, 2020-02-04:
I found a bug in data check.
when I loaded an urdf model that only include a box like this:

```
<robot
  name="block">
  <link name="block">

    <inertial>
      <origin xyz="0 0 0" />
      <mass value="0.1" />
      <inertia  ixx="0.00003375" ixy="0.0"  ixz="0.0"  iyy="0.00003375"  iyz="0.0"  izz="0.00003375" />
    </inertial>

    <visual>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.2 0.2 0.8" />
      </geometry>
    </visual>

    <collision>
      <origin xyz="0 0 0"/>
      <geometry>
        <box size="0.2 0.2 0.8" />
      </geometry>
    </collision>
  </link>

  <link name="world"/>
  <joint name="world-block-fixed" type="fixed">
    <parent link="world"/>
    <child link="block"/>
    <origin rpy="0 0 0" xyz="0.4 0.4 0.4" />
  </joint>
</robot>
```

after I created model data and gemotry data,  I used  updateGeometryPlacements and met an error  "data is not consistent with model."
data.kinematic_hessians.dimension(1) is not equal with model.nv.
Could you fix this bug?

```
pinocchio::Data env_data(env_model);
pinocchio::GeometryData env_gemo_data(env_geom_model);
Eigen::VectorXd env_q = randomConfiguration(env_model);
updateGeometryPlacements(env_model,env_data,space>env_geom_model,env_gemo_data,env_q);
```

jcarpent, 2020-02-04:
Dear @Linjackffy,

This bug is fixed on the devel branch of Pinocchio. Can you compile it on your side and provide us feedback. 

PS : This bug is coming in fact from Eigen 3.0-beta, provided by Ubuntu 16.04.

jmirabel, 2020-02-04:
For reference: duplicates https://github.com/stack-of-tasks/pinocchio/issues/1011.

jcarpent, 2020-02-04:
I will close this issue as it seems to be solved.

closed 2020-02-04 by jcarpent