stack-of-tasks/pinocchio #1056: pinocchio ros-kinetic version

mrsp, 2020-02-04:
Hello 

Did any updates happened to the ros-kinetic version of pinocchio ?

The version you get by:

` sudo apt-get install ros-kinetic-pinocchio 
`
is not traceable by ros as a standard ros package: 

```
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "pinocchio" with
  any of the following names:

    pinocchioConfig.cmake
    pinocchio-config.cmake

  Add the installation prefix of "pinocchio" to CMAKE_PREFIX_PATH or set
  "pinocchio_DIR" to a directory containing one of the above files.  If
  "pinocchio" provides a separate development package or SDK, be sure it has
  been installed. 
```

and furthermore if you manage to fix this issue you get a missmatch in the ros boost version and pinocchio requirements:

e.g.
```

/opt/ros/kinetic/include/pinocchio/container/boost-container-limits.hpp:29:7: error: #error "BOOST_MPL_LIMIT_LIST_SIZE value is lower than the value of PINOCCHIO_BOOST_MPL_LIMIT_CONTAINER_SIZE"
     # error "BOOST_MPL_LIMIT_LIST_SIZE value is lower than the value of PINOCCHIO_BOOST_MPL_LIMIT_CONTAINER_SIZE"
       
```
any ideas how to surpass the latter issues within an easy way for users?

Finally what version of boost is preferred for pinocchio to properly function in 16.04?

Thanks in advance

jcarpent, 2020-02-04:
The main issue is that you should pass some compilation arguments that are not inside the pinocchioConfig.cmake

A quick fix for it is just to add the line to your CMake project:
```
add_definitions("-DBOOST_MPL_LIMIT_LIST_SIZE=30")

```

We are now working on having pinocchioConfig.cmake integrating these flags directly. This is planned for Pinocchio 2.3.0.

jcarpent, 2020-02-04:
I think @wxmerkt can provide you some guidance.

wxmerkt, 2020-02-04:
Hi @mrsp,
How are you trying to find the package? Using `catkin` or `cmake` or `pkg-config`? The version in Kinetic is 2.2.1 and I think we also need to have a look at your include order. It is imperative that you include `pinocchio/fwd.hpp` before any ros.h header due to the MPL limit sizes.

mrsp, 2020-02-04:
Hello again and thanks for the quick response

this is my CMakeLists

> https://github.com/mrsp/serow/blob/master/CMakeLists.txt

so its through the catkin, still this does not compile as it used to some time ago but rather you receive the error that pinocchio is not found.

Additionally, in my code #include <ros/ros.h> is the first thing included and that was the case for more than a year now.  I'll try to include the pinocchio headers first and provide more feedback.

thanks again

jcarpent, 2020-02-04:
`add_definitions("-DBOOST_MPL_LIMIT_LIST_SIZE=30")` should do the trick without changing your code.

mrsp, 2020-02-04:
ok adding the definitions and finding pinocchio through:

> 
> find_package(PkgConfig REQUIRED)
> pkg_check_modules(PINOCCHIO pinocchio REQUIRED)

Started the compilation now i get multiple errors about redefinitions for example:

> In file included from /opt/ros/kinetic/include/pinocchio/parsers/urdf.hpp:11:0,
>                  from /home/mrsp/Desktop/serow_ws/src/serow/include/serow/robotDyn.h:33,
>                  from /home/mrsp/Desktop/serow_ws/src/serow/include/serow/quadruped_ekf.h:36,
>                  from /home/mrsp/Desktop/serow_ws/src/serow/src/quadruped_ekf.cpp:34:
> /opt/ros/kinetic/include/pinocchio/parsers/urdf/types.hpp:45:32: error: redefinition of ‘template<class T, class U> boost::shared_ptr<X> urdf::const_pointer_cast(const boost::shared_ptr<U>&)’
>    PINOCCHIO_URDF_SHARED_PTR(T) const_pointer_cast(PINOCCHIO_URDF_SHARED_PTR(U) const & r)

what I do in the code is to parse a urdf file with:

> pinocchio::urdf::buildModel(model_name, *pmodel_, verbose);

any ideas on this?

Thanks again for the support

jcarpent, 2020-02-04:
You should use ` pkg-config --cflags pinocchio` to watch the missing flags.
In your case, you should add:
`-DPINOCCHIO_URDFDOM_TYPEDEF_SHARED_PTR -DPINOCCHIO_URDFDOM_USE_STD_SHARED_PTR -DBOOST_MPL_LIMIT_LIST_SIZE=30 -DPINOCCHIO_WITH_URDFDOM`

jcarpent, 2020-02-04:
@mrsp Sorry for the durty fix, but normally Pinocchio 2.3.0 will provide the fix.

wxmerkt, 2020-02-04:
There are actually two different fixes:

1. Use the CMake, not the catkin way to find it. We deprecated and removed the catkin configs in favour of regular CMake:  `find_package(pinocchio REQUIRED)`.
2. Add the cflags when using PkgConfig:

```
target_compile_definitions(${PROJECT_NAME} PRIVATE ${PINOCCHIO_CFLAGS_OTHER})
```

mrsp, 2020-02-04:
@jcarpent  and @wxmerkt many thanks for the fixes/quick responses i will properly update my repository tomorrow, try it in a clean ubuntu 16.04 ros-kinetic environment and let you know about it.  

In my work i use pinocchio built by source and never encountered such errors, nevertheless I suggested my users to get the ros pinocchio package since it was easier and faster to install. 

thanks again

mrsp, 2020-02-05:
Good morning @jcarpent  and @wxmerkt 

The issue is resolved with the cflags in PkgConfig everything functions fine once again!

Once again I appreciate the help!

jcarpent, 2020-02-05:
Thanks for the quick feedback. We can then consider the issue solved.
Enjoy !

closed 2020-02-05 by jcarpent

dvogureckiy99, 2023-02-21:
Hi, I have the same issue. 
When I try build another project with your library pinocchio I have error like illustrated above. cflags in PkgConfig doesn't solve my issue.

jcarpent, 2023-02-21:
Could you explain a bit more @dvogureckiy99?