stack-of-tasks/pinocchio #1057: memory allocation in pinocchio::integrate

hammoudbilal, 2020-02-05:
Hi, 
I am trying to improve the speed of some code depending on pinocchio, I am using 
Eigen::internal::set_is_malloc_allowed(false);
to disable memory allocation for eigen, it seems that the only function to allocate some memory is pinocchio::integrate(). Is there a way around it ?

jcarpent, 2020-02-05:
Could you provide a snipset of your code?

jcarpent, 2020-02-05:
Yes, in fact, you should call in fact:
```
integrate(model,q_in,v_in,q_out)
```

hammoudbilal, 2020-02-05:
sure, pinocchio::aba doesn't seem to create an issue. 

Eigen::internal::set_is_malloc_allowed(false);
pinocchio::aba(*model_, *data_, q_, v_, tau_);
dv_ = data_->ddq; 
vMean_ = v_ + dv_ * .5 * sub_dt;
vMeanDt_ = vMean_ * sub_dt;
qnext_ = pinocchio::integrate(*model_, q_, vMeanDt_);
v_ += data_->ddq * sub_dt; 
Eigen::internal::set_is_malloc_allowed(true);

hammoudbilal, 2020-02-05:
thanks alot !

closed 2020-02-05 by hammoudbilal