stack-of-tasks/pinocchio #1061: ABA timings with pyrene arm

proyan, 2020-02-06:
Hi Justin,

I'm checking the performance of ABA in crocoddyl, since there seems to be something off.
When I run the timings benchmark in pinocchio (which is for a 36 DoF system), I get the following times for ABA
`ABA = 	11.3696 us`

which is not great, but bearable in my computer.

Now I made a small program to test the performance with the talos arm (7 DOF system)
```
#include <iostream>

#include <pinocchio/parsers/urdf.hpp>
#include <pinocchio/algorithm/aba.hpp>
#include <example-robot-data/path.hpp>

#include "pinocchio/utils/timer.hpp"

int main(int argc, char* argv[]) {

  unsigned int T = 1000;
  
  PinocchioTicToc timer(PinocchioTicToc::US);
  
  pinocchio::Model model;
  pinocchio::urdf::buildModel(EXAMPLE_ROBOT_DATA_MODEL_DIR "/talos_data/robots/talos_left_arm.urdf", model);
  pinocchio::Data data(model);

  Eigen::VectorXd q0 = Eigen::VectorXd::Random(model.nq);
  Eigen::VectorXd v0 = Eigen::VectorXd::Random(model.nv);
  Eigen::VectorXd t0 = Eigen::VectorXd::Random(model.nv);

  std::cout<<"NQ = "<<model.nq<<std::endl;
  timer.tic();
  SMOOTH(T) {
    pinocchio::aba(model, data, q0, v0, t0);
  }
  std::cout << "ABA = \t"; timer.toc(std::cout,T);
}

```

The output of this is: 

```
NQ = 7
ABA = 	810.118 us
```

which is really bad.

Could you please help me figure out what is happening here?

olivier-stasse, 2020-02-07:
@proyan 
Thanks for the benchmarking. Looking at your code, IMHO you should remove std::cout << "ABA = \t" . The way it is written it looks that it is including in the timing. Given the fact that only 1000 iterations are done, and that stdout is notoriously time consuming, the stats are probably wired.
Finally looking at src/timer.hpp the method with toc(std::ostream,...) relies in the compiler to do everything in the right order, I would call gettimeofday before the << operator.
But it is probably old school...

olivier-stasse, 2020-02-07:
@proyan could you confirm that you are not in DEBUG mode ?
I just tried with the following code and in DEBUG mode I have 711 us and in RELEASE mode I got 2.3 us on a Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz.
```
#include <iostream>

#include <pinocchio/parsers/urdf.hpp>
#include <pinocchio/algorithm/aba.hpp>
#include <example-robot-data/path.hpp>

#include "pinocchio/utils/timer.hpp"

int main(int , char* []) {

  unsigned int T = 1000;
  
  PinocchioTicToc timer(PinocchioTicToc::US);
  
  pinocchio::Model model;
  pinocchio::urdf::buildModel(EXAMPLE_ROBOT_DATA_MODEL_DIR "/talos_data/robots/talos_left_arm.urdf", model);
  pinocchio::Data data(model);

  Eigen::VectorXd q0 = Eigen::VectorXd::Random(model.nq);
  Eigen::VectorXd v0 = Eigen::VectorXd::Random(model.nv);
  Eigen::VectorXd t0 = Eigen::VectorXd::Random(model.nv);

  std::cout<<"NQ = "<<model.nq<<std::endl;
  timer.tic();
  SMOOTH(T) {
    pinocchio::aba(model, data, q0, v0, t0);
  }
  double ldt = timer.toc(PinocchioTicToc::US)/(double)T;
  std::cout << "ABA = \t" << ldt << std::endl ;
}
```

jcarpent, 2020-02-07:
@proyan, I think @olivier-stasse is right and you aren’t using O3 flags when compiling.

cmastalli, 2020-02-07:
aba goes particularly slow in DEBUG mode (slower than forward dynamics).
It seems the difference is there.
In any case, it will help to know the compilation command that you're using it

proyan, 2020-02-07:
@olivier-stasse @jcarpent thanks for the quick help. You were right, I was under the misapprehension that RELEASE is the default compilation mode. It seems that not using -O3 is as slow as DEBUG, and without any debugging symbols.

I used -O3 in the flags, and now the time is ~3us. Thanks.

@cmastalli: I was simply compiling, and didn't use any optimization params.

I'll close this issue.

closed 2020-02-07 by proyan