stack-of-tasks/pinocchio #1062: Eigenpy not found at configuration time

vpadois, 2020-02-07:
Hello,

I am installing pinocchio from source in order to use the devel branch. I installed Eigenpy as indicated here : https://github.com/stack-of-tasks/eigenpy but when running cmake for pinocchio I get the following error : 
`-- eigenpy >= 1.5.0 is required.
-- Checking for module 'eigenpy>=1.5.0'
--   No package 'eigenpy' found
CMake Error at /home/vpadois/.local/lib/python2.7/site-packages/cmake/data/share/cmake-3.12/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /home/vpadois/.local/lib/python2.7/site-packages/cmake/data/share/cmake-3.12/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  cmake/pkg-config.cmake:311 (PKG_CHECK_MODULES)
  cmake/pkg-config.cmake:570 (ADD_DEPENDENCY)
  bindings/python/CMakeLists.txt:48 (ADD_COMPILE_DEPENDENCY)
`

Should I just tell cmake to ignore Eignepy or is there something smarter to do ?

Thanks in advance for your help.

Best
Vincent

florent-lamiraux, 2020-02-07:
Hello Vincent,

It seems that the detection of eigenpy by  pkgconfig fails. Can you check

1. that executable `pkg-config` is installed (and accessible by your `PATH`). Simply type
`pkg-config --version` in the terminal in which you run `make`,
2. that eigenpy is installed and that `eigenpy.pc` is in a directory listed in `PKG_CONFIG_PATH` ?
`pkg-config --modversion eigenpy` should return something

vpadois, 2020-02-07:
Hello Florent,

Thank you for your help. This is what I get : 

>     1. that executable `pkg-config` is installed (and accessible by your `PATH`). Simply type
>        `pkg-config --version` in the terminal in which you run `make`,

`vpadois@ychoux:~/Documents/Softs/pinocchio/build$ pkg-config --version
0.29.1`

> 
>     2. that eigenpy is installed and that `eigenpy.pc` is in a directory listed in `PKG_CONFIG_PATH` ?
>        `pkg-config --modversion eigenpy` should return something

`vpadois@ychoux:~/Documents/Softs/pinocchio/build$ pkg-config --modversion eigenpy
Package eigenpy was not found in the pkg-config search path.
Perhaps you should add the directory containing `eigenpy.pc'
to the PKG_CONFIG_PATH environment variable
No package 'eigenpy' found`

So there is clearly a PATH problem for eigenpy

florent-lamiraux, 2020-02-07:
It is rather a PKG_CONFIG_PATH. When you install several packages that depend on one another from source, you should define a few environment variables to detect the files installed in non-standard directories.
For instance, if you install in a prefix called `PREFIX`, you should add
- `${PREFIX}/lib/pkgconfig` in your `PKG_CONFIG_PATH`,
- `${PREFIX}/lib` in your `LD_LIBRARY_PATH`
- `${PREFIX}/bin` in your `PATH`,
- `${PREFIX}/lib/pythonx.y/site-packages:${PREFIX}/lib/pythonx.y/dist-packages` in your `PYTHONPATH`.
- ${PREFIX} in your `CMAKE_PREFIX_PATH`.

vpadois, 2020-02-07:
I understand that well but the point is that I did not use a specific PREFIX (I did not install eigenpy from source but using the proposed bionic package) so I did not expect that I would have some environment variable to define. Does this make sense or am I missing something ?

florent-lamiraux, 2020-02-07:
Unfortunately, binary packages from third party sources are not installed in standard directories:
- `robotpkg` installs in `/opt/openrobots`,
- `ros` installs in `/opt/ros/melodic` (for `ros-melodic` packages).
Therefore, you still need to set environment variables.

vpadois, 2020-02-07:
Ok. I will do that. Thank for your help.

closed 2020-02-07 by vpadois

jcarpent, 2020-02-07:
By the way, @vpadois you can set `BUILD_PYTHON_INTERFACE` to `OFF` when compile Pinocchio.

vpadois, 2020-02-07:
OK. Will do. Thx