stack-of-tasks/pinocchio #1064: aba-derivatives without aba

proyan, 2020-02-07:
Hey Justin,
Would there be an appetite for a function in pinocchio which computes the aba-derivatives, and assumes that aba has already been called?

jcarpent, 2020-02-07:
Yes, I have it in mind indeed. But, the problem is that classic ABA is using local coordinates to representent the physical quantities, while the derivatibes are using global coordinates systems for efficiency.

jcarpent, 2020-02-07:
So, I first need to bench the current ABA to know if it worthy or not.

jcarpent, 2020-02-07:
But, anyway, you won't gain so much. What is costly in ABA is Minv.

proyan, 2020-02-07:
Do you know if and when this might be available?

> But, anyway, you won't gain so much. What is costly in ABA is Minv.

Only ABA might not be that much, but ABA is running in a serial manner multiple times.  So what I hope to gain is not simply t(aba) but t(aba)*number_of_nodes.

jcarpent, 2020-02-08:
The gain you can expect is something around 3 micro per node, no more. Is it really worthy?

jcarpent, 2020-02-08:
In addition, I really advise you to not use your computer as a reference for timings. You have indeed a pessimist configuration.

proyan, 2020-02-10:
> In addition, I really advise you to not use your computer as a reference for timings. You have indeed a pessimist configuration.

That is true, unfortunately :(

However, regarding this issue, I can spend a day on it later this week if you have constraints. Depending on how much we can improve multithreading, I think we could have a benefit of 50-100 us in the backward pass of DDP.

jcarpent, 2020-02-10:
I'm currently evaluating the benefits of such tricks. I will let you know the gains if any.

jcarpent, 2020-02-12:
@proyan Just to tell you that I've done the bench, and it seems that reformulating ABA in the world coordinates system is decreasing the performances of 0.5 us on 36-dof humanoid robot.
So, it seems reasonable to integrate this feature for the analytical derivatives of ABA to avoid additional computations of the joint vector acceleration if the computations have already been performed. 

When will you need this feature?

proyan, 2020-02-12:
The sooner the better. I'm not blocked by this, since I have some other avenues of improvement as well. But I can integrate this in crocoddyl as soon as it is available.

proyan, 2020-02-13:
> @proyan Just to tell you that I've done the bench, and it seems that reformulating ABA in the world coordinates system is decreasing the performances of 0.5 us on 36-dof humanoid robot.

This is more a comment on crocoddyl than Pinocchio, but I think I could extend this performance bump you saw in ABA to other feature computations as well. For e.g, when calculating the frame position and jacobians, LOCAL frame is much more costly that WORLD frame. I'll try to benchmark this as well and let you know if there is any improvement.

jmirabel, 2020-02-13:
> reformulating ABA in the world coordinates system is decreasing the performances of 0.5 us on 36-dof humanoid robot.

@jcarpent could you precise the following please ?
- wall time or CPU time ?
- 0.5us is the order of magnitude of a few (3-5) *worst case* cache misses. I would be curious to have both wall time and cpu time to see how this time improvement is shared.
- what is the total time of either version (to know what percentage this represents) ?

jcarpent, 2020-02-13:
According to the bench in Pinocchio, the bench only measures the wall time, not the CPU time of a given algorithm.

Concerning ABA, the current ABA takes 5.70 us on my i7 Core @ 2.5 Ghz without hyperthreading against 6.12 us for the new version.

We would need to update the current bench to really measure the CPU time, this should be something really helpful I think.

jmirabel, 2020-02-13:
Thanks for the clarifications.

jcarpent, 2022-05-04:
Done in pinocchio-3x which should be release soon.

closed 2022-05-04 by jcarpent