stack-of-tasks/pinocchio #1065: Compatibility issue

dadouz, 2020-02-11:
Hi,
i m trying to install pinocchio python library on a raspberryPi4.
The system is blocking the installation and says that the architecture (aarch64) is not supported at following stage: sudo apt install robotpkg-py27-pinocchio
On a linux virtual machine (architecture x86_64) installed on Windows OS, everythimng works fine.

Can you provide me with a workaround to solve this problem on the RaspberryPi4.
We are using Ubuntu 18.4.3 TLS on the RaspPi4.

Thanks in advance.

jcarpent, 2020-02-11:
Dear @dadouz,

In short, current binairies are only available for amd64 architecture (not ARM). To install it on another architecture, I would advise you to rely on the following instructions to install all the dependencies using either robotpkg (https://stack-of-tasks.github.io/pinocchio/download.html#Install_3) or directly from sources (https://stack-of-tasks.github.io/pinocchio/download.html#Install_4).

Best,

Justin

wxmerkt, 2020-02-11:
The binaries provided via the ROS buildfarm should support arm

jcarpent, 2020-02-13:
@dadouz Any feedback?

jcarpent, 2020-02-18:
I will close this issue hopping that the problem is now solved.

closed 2020-02-18 by jcarpent

Linjackffy, 2020-04-02:
I also tried to install pinocchio on an arm-based development board. 
The OS run on the borad is Linux 4.1.15.
How could I get binaries provided via the ROS buildfarm?