stack-of-tasks/pinocchio #1066: [bionic 18.04] robotpkg-osg-dae and robotpkg-ur5-description does not exist

mohamdev, 2020-02-12:
Hello everyone, 

Im posting this issue because i'm trying to view my robot on gepetto-gui. I get an error which says that it is impossible to read .dae files and the latter should be converted in .osg. I tried to install robotpkg-osg-dae, but this package doesn't exist in robotpkg bionic (I checked on this link : http://robotpkg.openrobots.org/packages/debian/pub/dists/bionic/robotpkg/binary-amd64/). I had the same issue for installing robotpkg-ur5-description.

To avoid this problem, I added the xenial version of robotpkg (instead of bionic, which is my ubuntu version) and then tried to install both packages, using these commands : 

- sudo sh -c "echo 'deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub **xenial** robotpkg' >> /etc/apt/sources.list.d/robotpkg.list"

- curl http://robotpkg.openrobots.org/packages/debian/robotpkg.key | sudo apt-key add -

- sudo apt-get update

- sudo apt install robotpkg-ur5-description

- sudo apt install robotpkg-osg-dae

This works fine for the ur5-description package which gets installed. But, the robotpkg-osg-dae can't get installed, I get the following error : 

_Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 robotpkg-osg-dae : Depends: libboost-filesystem1.58-dev (>= 1.34.1) but it is not installable
                    Depends: libboost-filesystem1.58.0 (>= 1.34.1) but it is not installable
                    Depends: libboost-system1.58-dev (>= 1.34.1) but it is not installable
                    Depends: libboost-system1.58.0 (>= 1.34.1) but it is not installable
E: Unable to correct problems, you have held broken packages._

After that, I installed libboost-all-dev, but it actually changes nothing. Thank you for your feedback

jcarpent, 2020-02-12:
A quick answer to your issue is that you have to add robotpkg-wip to the list of known servers. See http://robotpkg.openrobots.org/robotpkg-wip.html for further details.

PS: for information, wip packages are not totally stable but almost ready for production.
@nim65s We should move osg-dae to the main robotpkg repository I think.

mohamdev, 2020-02-12:
Thank you very much, it works fine with wip :)
I followed the instructions there : https://github.com/stack-of-tasks/pinocchio/issues/705#issuecomment-467835859

jcarpent, 2020-02-12:
Great. I will close this issue then.

closed 2020-02-12 by jcarpent

perak96, 2020-02-13:
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
List of GroupNodes :
gepetto-gui
hpp-gui
python-pinocchio
BOX does not exist.: iostream error
BOX does not exist.: iostream error
BOX does not exist.: iostream error
BOX does not exist.: iostream error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.
You must initialize a Geode on world/pinocchio/visuals/base_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/shoulder_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/upper_arm_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/forearm_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/wrist_1_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/wrist_2_link_0 to use Alpha
You must initialize a Geode on world/pinocchio/visuals/wrist_3_link_0 to use Alpha
BOX does not exist.: iostream error
BOX does not exist.: iostream error

When I try to visualize robot ur5.urdf in gepetto-gui, I get this message and only one segment is showed in the visualization screen. Can somebody explain what is happening and how I can fix this?

Thank you in advance.

jcarpent, 2020-02-14:
Can you share your code to see what is missing. By the way, in the folder examples, you have an example of using Pinocchio with the viewer.

jmirabel, 2020-02-14:
Line *BOX does not exist.: iostream error* means a) you are not up to date and b) that you have a specific UR5 model because I think there is no box in the official UR5 model.

perak96, 2020-02-14:
@jcarpent 
This is my code( I followed the example):
from pinocchio.robot_wrapper import RobotWrapper
import pinocchio
import gepetto
import numpy
pinocchio.switchToNumpyMatrix()

URDF = '/opt/openrobots/share/ur5_description/urdf/ur5_gripper.urdf'

robot = RobotWrapper.BuildFromURDF(URDF)
robot.initViewer(loadModel = True)
#rgbt = [0.0, 0.0, 1.0, 1.0]  # red, green, blue, transparency
#robot.viewer.gui.addSphere("world/sphere", .1, rgbt)  # .1 is the radius

visualObj = robot.visual_model.geometryObjects[4]  # 3D object representing the robot forarm
visualName = visualObj.name                        # Name associated to this object
visualRef = robot.getViewerNodeName(visualObj, pinocchio.GeometryType.VISUAL)  

rgbt = [0.0, 0.0, 1.0, 1.0]  # red, green, blue, transparency
q1 = (.5, .1, .2, 1.,0.,0.,0. )
robot.viewer.gui.addSphere("world/sphere", .1, rgbt)  # .1 is the radius
robot.viewer.gui.applyConfiguration("world/sphere", q1)
robot.viewer.gui.refresh()
This is the photo of a gepetto-gui when I introduce a robot in it. Should it look like this? It seems that only the meshes can be drawn in gui.
[image]

jcarpent, 2020-02-14:
you can do `robot.display(robot.q0)` to display the neutral configuration. You will see the whole robot then.

perak96, 2020-02-14:
It worked. One more question: Although I see seven links, I get this message:
BOX does not exist.: iostream error
BOX does not exist.: iostream error
BOX does not exist.: iostream error
BOX does not exist.: iostream error
Node "world/pinocchio/collisions/ee_link_0" does not exist.
Node "world/pinocchio/collisions/tool0_0" does not exist.
Node "world/pinocchio/collisions/tool0_1" does not exist.
Node "world/pinocchio/collisions/tool0_2" does not exist.

Is it true that only meshes can be shown in gepetto gui? My robot now looks like this:
[image]