stack-of-tasks/pinocchio #1069: mobilerobot source code unavailable

mohamdev, 2020-02-16:
Hello, i'm trying to follow the tutorial "Drag and drop (aka inverse kinematics)", but I can't access to the mobilerobot module. In fact, in the tutorial, it is specified that the module is available on this link : https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/mobilerobot_8py_source.html

But when I click on the link I have a 404 error. I guess I can't perform the tutorial witheout this module. Thank you for your feedback :)

jcarpent, 2020-02-16:
@gabrielebndn I think this issue issue is coming from your side. Could you provide some help please?

gabrielebndn, 2020-02-16:
@jcarpent he is actually referring to https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/md_doc_d-practical-exercises_3-invkine.html, which is not my work. There is currently some confusion and overlap between the different tutorials and examples etc.

The link is inside the webpage, but it is currently not working. I think the best person to answer this is @nmansard. Either him or @nim65s

jcarpent, 2020-02-16:
@gabrielebndn I found the bug and I think you did want it, but it seems your made the modifications that lead to the current behavior ;) Please see https://github.com/stack-of-tasks/pinocchio/commit/68bed7b27b95e9e9569474bf02a4b420309cfd8f
I will provide a fix.

gabrielebndn, 2020-02-16:
Oooh, I see... I will comment on the PR.
Speaking of which, wouldn't it be time to manually update the uploaded doc, at least partially addressing #1007 before an automatic solution is put back in place?

gabrielebndn, 2020-02-16:
Meanwhile, @mohamdev, while waiting for the fix, all Python sources for the preactical exercises can be found in the Pinocchio repo, in the subfolder [doc/d-practical-exercises/src](https://github.com/stack-of-tasks/pinocchio/tree/e038c7bf283b1df56a35014455e0e2d6f36e03ac/doc/d-practical-exercises/src). You can either copy and paste the files one by one from Github when you need them, or you can clone the repo. This is the most convenient solution, I think: you don't even need to build it, just clone the repo and and then copy the files somewhere else

jcarpent, 2020-02-17:
Solved by #1071.
@mohamdev We will provide a new release of the doc soon. Thanks a lot for raising this very helpful issue.

closed 2020-02-17 by jcarpent

mohamdev, 2020-02-17:
That's fine, thank you very much for your feedback :)

PR #1071 (2020-02-17): [doc] [practical exercises] Fix file links
Fixes #1069. Alternative to #1070