stack-of-tasks/pinocchio #1072: Linking to header files in CMakeLists.txt

Neotriple, 2020-02-18:
Hey all,

I'm actually trying to link to the header files that I've installed in a directory which I'll call:
pinocchio_DIR

However, when I try to build a simple test application to load a URDF, and do a 'make' on the folder, I run into this issue:

```
/home/pshah/pinocchio_ws/src/pinocchio/src/parsers/urdf.hpp:9:10: fatal error: pinocchio/multibody/model.hpp: No such file or directory
 #include "pinocchio/multibody/model.hpp"
```

For reference, my CMakeList.txt looks something like this:

```
cmake_minimum_required(VERSION 3.15)
project(untitled)
find_package(Eigen3 REQUIRED)
find_package(ifopt REQUIRED)

set(CMAKE_CXX_STANDARD 17)
set(pinocchio_DIR "link/to/pinocchio/src")

include_directories(
        ${pinocchio_DIR}
)

add_executable(untitled main.cpp)

target_link_libraries(untitled
    PUBLIC
        Eigen3::Eigen
        ifopt::ifopt_core
        ifopt::ifopt_ipopt
)
```

I'm sure I'm just doing something quite stupid, but I can't quite figure it out. I have indeed added the path environments to my .bashrc (as suggested in the 'build from source' section), but I don't think this has anything to do with it since I'm trying to directly link to the folders.

jcarpent, 2020-02-18:
@Neotriple If can pull the devel branch of Pinocchio and then add to your CMakeLists.txt these two lines:
```
find_package(pinocchio REQUIRED)
```
and:
```
target_link_libraries(untitled
    PUBLIC
        Eigen3::Eigen
        ifopt::ifopt_core
        ifopt::ifopt_ipopt
        pinocchio::pinocchio
)
```

jcarpent, 2020-02-19:
@Neotriple Any news?

Neotriple, 2020-02-19:
Sorry for the lack of update. I forgot to mention that I'm building using the catkin build toolset. When I went to rebuild after switching to the devel branch, the catkin build toolset failed due to an error in eigenpy. Once I figure out this error, I can try your method above.

As a general question, the documentation notes that pinocchio is now header-only, however, from my understanding, find_package finds libraries by searching through the CMAKE_MODULE_PATH, however since pinocchio is now header-only, doesn't this mean there is no library to be made? Or are the 'headers' essentially the library?

wxmerkt, 2020-02-19:
@Neotriple The CMake find_package won't work for you yet. I'll come find you to debug

jcarpent, 2020-02-19:
@wxmerkt Could you update ROS with the new release of Pinicchio with full cmake package support now.

wxmerkt, 2020-02-19:
I've made the releases. Thank you very much for the updates and CMake package support :-).

jcarpent, 2020-02-19:
@wxmerkt Thanks a lot for the release.

jcarpent, 2020-02-20:
@wxmerkt Could update ROS with the new release?

wxmerkt, 2020-02-20:
Done :-) 

I will close the issue as Paarth and I debugged this in person yesterday.

closed 2020-02-20 by wxmerkt

jcarpent, 2020-02-20:
@wxmerkt Thanks a lot.