stack-of-tasks/pinocchio #1073: Drag and drop (Inverse kinematics) issues

mohamdev, 2020-02-19:
Hello :),

I have some issues performing the Drag and drop tutorial. 

1) When I try to import the URDF of the mobile ur5 using the command 

`robot = MobileRobotWrapper(URDF, [PKG])`

I get the following error : 

> AttributeError: 'MobileRobotWrapper' object has no attribute 'createData'

I tried to use the command `MobileRobotWrapper.BuildFromURDF(URDF)`, it works fine, but it imports the 6 DOF ur5 (robot.nq = 6), yet the mobile version of ur5 should be 9 DOF (robot.nq should be equal to 9), so I guess it imported it wrong.

2) In the same tutorial there are these two commands : 
```

IDX_TOOL = 24
IDX_BASIS = 23
```
How can be indexes equal to 24 and 23 as the robot has only 9 DOF ? 

3) In the same tutorial, there is this command `se3.framesKinematics(robot.model, robot.data)`, but when I tip it, I have the following error : 

> AttributeError: 'module' object has no attribute 'framesKinematics'

Thank you very much for your feedback :)

gabrielebndn, 2020-02-19:
> Hello :),
> 
> I have some issues performing the Drag and drop (aka inverse kinematics) tutorial.
> 
>     1. When I try to import the URDF of the mobile ur5 using the command
> 
> 
> `robot = MobileRobotWrapper(URDF, [PKG])`
> 
> I get the following error :
> 
> > AttributeError: 'MobileRobotWrapper' object has no attribute 'createData'
> 
> I tried to use the command `MobileRobotWrapper.BuildFromURDF(URDF)`, it works fine

Indeed, the tutorials are very out of date. `BuildFromURDF` is the current method.

but it imports the 6 DOF ur5 (robot.nq = 6), yet the mobile version of ur5 should be 9 DOF (robot.nq should be equal to 9), so I guess it imported it wrong.

That's because if the base is not fixed the root joint must be provided externally (this info is not contained in the URDF, thus there is no way for Pinocchio to retrieve it. In your case, you need a `JointModelPlanar`:

```
`MobileRobotWrapper.BuildFromURDF(URDF,  pkgs, pin.JointModelPlanar())`
```

The [custom constructor](https://github.com/stack-of-tasks/pinocchio/blob/7765909f1aed41dd9471baaf8e492be5fcc99523/doc/d-practical-exercises/src/mobilerobot.py#L19-L36) of `MobileRobotWrapper` was taking care of this for you, but you could not use it because `MobileRobotWrapper` employs an old interface.
Notice that the constructor of `MobileRobotWrapper` was also doing other stuff, such as adding custom frames, which might be expected in the follow-up of the tutorials.

Therefore, you can either take a look at the constructor of `MobileRobotWrapper` and repeat its same steps, or fix `MobileRobotWrapper` itself. In the latter case, the best thing would be to rework the class so that its interface complies more to that of `RobotWrapper`, but I think a quick fix for you would be simply to replace this line

https://github.com/stack-of-tasks/pinocchio/blob/7765909f1aed41dd9471baaf8e492be5fcc99523/doc/d-practical-exercises/src/mobilerobot.py#L20

with
```
self.initFromURDF(urdf, pkgs, pin.JointModelPlanar())
```

>     1. In the same tutorial there are these two commands :
> 
> 
> ```
> 
> IDX_TOOL = 24
> IDX_BASIS = 23
> ```
> 
> How can indexes be equal to 24 and 23 as the robot has only 9 DOF ?

That's because those are not indexes of _links_, but of _frames_. One link can have an arbitrary number of frames attached to it (including zero). You can take a look [in the doc](https://gepettoweb.laas.fr/doc/stack-of-tasks/pinocchio/master/doxygen-html/structpinocchio_1_1FrameTpl.html)

>     1. In the same tutorial, there is this command `se3.framesKinematics(robot.model, robot.data)`, but when I tip it, I have the following error :
> 
> 
> > AttributeError: 'module' object has no attribute 'framesKinematics'
> 
> Thank you very much for your feedback :)

Again, the tutorials are out of date. The command would be `updateFramePlacements(robot.model, robot.data)`, or `framesForwardKinematics(robot.model, robot.data, q)` if you contextually provide `q`

jcarpent, 2020-02-19:
@gabrielebndn I think you have a correct understanding of what is missing now in the tutorial to be up to date. It would be great if you can correct them in such a way nobody will be annoyed. Can you take it in charge?

gabrielebndn, 2020-02-19:
I will see what I can do, but I am full of work, so it will be very low priority

gabrielebndn, 2020-02-19:
Mostly, I think we should find a way to unit test them

jcarpent, 2020-02-19:
Yes, sure for the CI.

jcarpent, 2020-02-19:
From my side, I'm also full and I think you might be much more efficient than me on it. @nmansard @nim65s Do you think that you have some times to fix it?

nim65s, 2020-02-19:
@nmansard created new tutorials. He is waiting for me to review them. I am planning to do that in a week or 2.
This will come with proper CI integration.

gabrielebndn, 2020-02-19:
Excellent then!

mohamdev, 2020-02-19:
@gabrielebndn Thank you very much for your feedback, it was really helpful :) 
I'm glad to hear that tutorials will be updated. Thank you for your helpful work

closed 2020-02-22 by mohamdev