stack-of-tasks/pinocchio #1075: [Python] exception handling issue

nim65s, 2020-02-19:
Hi,

https://github.com/stack-of-tasks/pinocchio/blob/7765909f1aed41dd9471baaf8e492be5fcc99523/examples/display-shapes.py#L40

On my system, this leads to the following error:
```
Error while initializing the viewer. It seems you should install gepetto-viewer
Traceback (most recent call last):
  File "…/pinocchio-2.3.0/examples/display-shapes.py", line 37, in <module>
    viz.initViewer()
  File "…/pinocchio-2.3.0/bindings/python/pinocchio/visualize/gepetto_visualizer.py", line 28, in initViewer
    import gepetto.corbaserver
ModuleNotFoundError: No module named 'gepetto'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "…/pinocchio-2.3.0/examples/display-shapes.py", line 40, in <module>
    print(error.message)
AttributeError: 'ModuleNotFoundError' object has no attribute 'message'
```

I can probably handle this if no one else is available.

jcarpent, 2020-02-19:
Please go ahead with the fix. Thanks in advance @nim65s.

jcarpent, 2020-02-20:
@nim65s I plan to make a new release with bug fixes. Could you handle this issue today?

jcarpent, 2020-02-20:
In fact, I will handle it by myself

jcarpent, 2020-02-20:
Solved by #1081.

closed 2020-02-20 by jcarpent