stack-of-tasks/pinocchio #1076: [Tests] Missing link to urdfdom

nim65s, 2020-02-19:
Hi,

https://github.com/stack-of-tasks/pinocchio/blob/7765909f1aed41dd9471baaf8e492be5fcc99523/unittest/multiprecision.cpp#L75

On my system, this code lacks link to urdfdom:
```
/usr/bin/ld: CMakeFiles/test-cpp-multiprecision.dir/multiprecision.cpp.o: undefined reference to symbol '_ZN4urdf13parseURDFFileERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE'
/usr/bin/ld: /usr/lib/liburdfdom_model.so.1.0: error adding symbols: DSO missing from command line
```
(demangled: `urdf::parseURDFFile()`)

This can be fixed there :
https://github.com/stack-of-tasks/pinocchio/blob/7765909f1aed41dd9471baaf8e492be5fcc99523/unittest/CMakeLists.txt#L125

with
```
 # Multiprecision arithmetic
-ADD_PINOCCHIO_UNIT_TEST(multiprecision)
+ADD_PINOCCHIO_UNIT_TEST(multiprecision urdfdom)
 ```

Am I missing anything, or should I just open a PR for this ?

jcarpent, 2020-02-19:
I've a fix for it.

jcarpent, 2020-02-19:
Solved by #1077.

closed 2020-02-19 by jcarpent