stack-of-tasks/pinocchio #1078: Compiling URDF and SRDF parsers

jcarpent, 2020-02-19:
Current implementations of URDF and SRDF parsers are header-only, meaning that the private flags related to the various versions of URDFDOM are transferred for correct compilation. In addition, for each new executable, we have to compile the signatures again and again, while by default, URDF and SRDF assumes double scalar type.

I would like to pass these parsers to the existing but empty Pinocchio. This lib will only contain the parsers signatures, nothing more. @jmirabel @wxmerkt @cmastalli @nim65s @nmansard @aelkhour 
What is your opinion on this suggestion?

jmirabel, 2020-02-20:
I like the idea but I fear it is gonna be hard to achieve. In HPP, we are using a different JointModelCollection. I think this would be an issue, right ?

jcarpent, 2020-02-20:
No, because the JointCollection is also in the signature of default Model in Pinocchio. 
But, this means that we still need to export the compilations flags.

jcarpent, 2020-02-20:
Another solution would be to proceed to a cast.

jmirabel, 2020-02-20:
> No, because the JointCollection is also in the signature of default Model in Pinocchio.

That's the point. We don't use the default Model signature.

jmirabel, 2020-02-20:
We could also use a visitor design pattern.

jcarpent, 2020-02-20:
What do you mean?

jmirabel, 2020-02-20:
Add a class UrdfVisitor that has a prototype such as:
```c++
class UrdfVisitor {
virtual void addFreeFlyer(/* relevant args, excluding the model */) = 0;
virtual void addRevolute(/* relevant args, excluding the model */) = 0;
// and one more for each type of joint

virtual void addLink(/* relevant args, excluding the model */) = 0;
// and any other method that should be exposed.
};
```

Then a child class
```
template <typename Model> ChildClass
{
Model& model;
// implementation of each of the parent virtual methods.
}
```

Then, the URDF parser becomes independant of class Model.

jcarpent, 2020-02-20:
My initial goal would be to not have to export URDF related flags when compiling Pinocchio. I don't see how it will happen in your case?

jmirabel, 2020-02-20:
In the library, you will have a code that depends on URDF related flags like this:
```c++
void parse(std::string filename, UrdfVisitor* visitor)
{
UrdfTree urdf (filename);
...
// Add a joint
visitor->addRevolute (position_in_parent, axis, limits...);
}
```
Then, the UrdfVisitor::addRevolute function will not depend on anything related to urdfdom. You should not need to include any urdfdom header.

jcarpent, 2020-02-20:
OK. Thanks for the clarification. As you seems much more aware on the topic, would you have time to implement it?

jmirabel, 2020-02-20:
I can put this onto my task list but it won't have a high priority. There is no hurry for this, is there ?

jcarpent, 2020-02-20:
No, of course.

jcarpent, 2020-02-20:
Thanks a lot @jmirabel for handling this feature.

jcarpent, 2020-03-09:
There is still the SRDF parser to do.

jmirabel, 2020-03-09:
You are right. I though the intend was to get rid of *urdfdom* stuff in the installed includes. The SRDF parser only relies on boost, and not on *urdfdom*.

jcarpent, 2021-08-23:
@nim65s or @proyan Could you handle this (SRDF)?

proyan, 2021-08-23:
This issue is in liaison with https://github.com/stack-of-tasks/pinocchio/issues/1498, and normally this should be resolved when 1498 is handled

jcarpent, 2022-05-04:
@proyan Could you handle this issue in pinocchio-3x?