stack-of-tasks/pinocchio #1082: No available formula with the name "eigenpy-python3"

ialong, 2020-02-20:
Running `brew install pinocchio` currently fails due to eigenpy now using python3 by default (there is no `eigenpy-python3` formula anymore).

This is the output:

```❯ brew install pinocchio
==> Installing pinocchio from gepetto/gepetto
Error: No available formula with the name "eigenpy-python3" (dependency of gepetto/gepetto/pinocchio)
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.```

jcarpent, 2020-02-20:
@ialong I'm currently regenerating the images. In 10 minutes, it should be available.

ialong, 2020-02-20:
great thanks! (amazing response speed :))

jcarpent, 2020-02-20:
Done! The binaries are now available.
Best,
Justin

closed 2020-02-20 by jcarpent

ialong, 2020-02-20:
It seems I'm still getting the same issue (after running brew update)...

jcarpent, 2020-02-20:
I would advise you to proceed as follows:
```
brew untap gepetto/geptto
brew tap gepetto/geptto
brew install pinocchio
```
or, for Python 2:
```
brew install pinocchio@2
```

ialong, 2020-02-20:
@jcarpent unfortunately still the same error.

jcarpent, 2020-02-20:
OK. I've found the bug.

jcarpent, 2020-02-20:
I just need 5 mins to solve it and regenerate the correct image

jcarpent, 2020-02-20:
Done. Thanks @ialong for your quick feedback.

ialong, 2020-02-20:
It worked! @jcarpent thank you very much for fixing this so quickly

jcarpent, 2020-02-20:
You're welcome.