stack-of-tasks/pinocchio #1089: Frame tree

jmirabel, 2020-02-24:
Hi,

I think there is a bug in the URDF parser. Consider the following URDF and python script.
```python
from __future__ import print_function
import pinocchio
pinocchio.switchToNumpyArray()

model = pinocchio.buildModelFromXML("""
<?xml version="1.0" encoding="utf-8"?>
<robot name="test">
  <link name="base_link"/>
  <link name="link_1"/>
  <link name="link_2"/>

  <joint name="joint_1" type="fixed">
    <origin xyz="1 0 0"/>
    <axis xyz="0 0 1"/>
    <!-- <limit lower="0" upper="0.001" effort="100" velocity="0.01"/> -->
    <parent link="base_link"/>
    <child link="link_1"/>
  </joint>
  <joint name="joint_2" type="fixed">
    <origin xyz="0 1 0"/>
    <axis xyz="0 0 1"/>
    <!-- <limit lower="0" upper="0.001" effort="100" velocity="0.01"/> -->
    <parent link="link_1"/>
    <child link="link_2"/>
  </joint>
</robot>
""")

for i, f in enumerate(model.frames):
    print("{}: id {}, parent {}".format(f.name, i, f.previousFrame))
```
It prints
```
universe: id 0, parent 0
root_joint: id 1, parent 0
base_link: id 2, parent 1
joint_1: id 3, parent 1
link_1: id 4, parent 3
joint_2: id 5, parent 3
link_2: id 6, parent 5
```

Clearly, the parent frame of joint_2 should be link_1 and not joint_1. Do we agree ?

jcarpent, 2020-02-24:
You're right I think too. Indeed, the tree of Frames should keep the URDF tree ordering.

jcarpent, 2020-02-24:
By the way, the same applies for joint_1 parent should be the base_link too.

jcarpent, 2020-02-24:
@jmirabel Can you handle the fix?

jmirabel, 2020-02-25:
I will put that with #1078

jcarpent, 2020-02-25:
I will then close it. Thanks for handling it.

closed 2020-02-25 by jcarpent