stack-of-tasks/pinocchio #1090: Pinocchio `getNumpyType` is not found in `def zero`

cmastalli, 2020-02-24:
Hi guys,

I have installed the latest devel branch of Pinocchio.
And when I run the following commands
```python
import pinocchio

vec = pinocchio.utils.zero(3)
```
I got the following error message:
```bash
/usr/local/lib/python2.7/dist-packages/pinocchio/utils.py in zero(n)
     38 
     39 def zero(n):
---> 40     if pin.getNumpyType()==np.matrix:
     41         return np.matrix(np.zeros([n, 1] if isinstance(n, int) else n))
     42     else:
```

Could it be that I need to update EigenPy and Pinocchio is not detecting this?

cmastalli, 2020-02-24:
**Update**
When I compiled the master branch of EigenPy everything works well.

The cmake needs to be updated in order to request for the right version of EigenPy. Unfortunately, I don't know which one is the minimal one.

jcarpent, 2020-02-24:
What is your old version of EigenPy?

cmastalli, 2020-02-24:
> What is your old version of EigenPy?

I think that was v1.6.1, however I am not sure about it.

jcarpent, 2020-02-24:
Solved by #1091.

closed 2020-02-24 by jcarpent